Laxilef/OTGateway #93: 1.4.6 errors and warnings

Pajuris, 2024-11-16:
Hello

After update to 1.4.6 version, appears error anw warning.

Error parsing value: 'dict object' has no attribute 'maxModulation' (value: {"heating":{"enable":true,"turbo":false,"target":21.5,"hysteresis":0.5,"turboFactor":7.5,"minTemp":20,"maxTemp":43},"dhw":{"enable":true,"target":45,"minTemp":30,"maxTemp":46},"equitherm":{"enable":false,"n_factor":0.7,"k_factor":3,"t_factor":2},"pid":{"enable":true,"p_factor":50,"i_factor":0.006,"d_factor":10000,"dt":180,"minTemp":5,"maxTemp":80},"sensors":{"outdoor":{"type":1,"gpio":12,"bleAddress":"00:00:00:00:00:00","offset":0},"indoor":{"type":1,"gpio":18,"bleAddress":"00:00:00:00:00:00","offset":0}}}, template: {{ value_json.heating.maxModulation|int(1) }})
Error parsing value: 'dict object' has no attribute 'maxModulation' (value: {"heating":{"enable":true,"turbo":false,"target":22,"hysteresis":0.5,"turboFactor":7.5,"Payload '{"heating":{"enable":true,"turbo":false,"target":21.5,"hysteresis":0.5,"turboFactor":7.5,"minTemp":20,"maxTemp":43},"dhw":{"enable":true,"target":45,"minTemp":30,"maxTemp":46},"equitherm":{"enable":false,"n_factor":0.7,"k_factor":3,"t_factor":2},"pid":{"enable":true,"p_factor":50,"i_factor":0.006,"d_factor":10000,"dt":180,"minTemp":5,"maxTemp":80},"sensors":{"outdoor":{"type":1,"gpio":12,"bleAddress":"00:00:00:00:00:00","offset":0},"indoor":{"type":1,"gpio":18,"bleAddress":"00:00:00:00:00:00","offset":0}}}' is not a Number
Payload '{"heating":{"enable":true,"turbo":false,"target":22,"hysteresis":0.5,"turboFactor":7.5,"minTemp":20,"maxTemp":43},"dhw":{"enable":true,"target":45,"minTemp":30,"maxTemp":46},"equitherm":{"enable":false,"n_factor":0.7,"k_factor":3,"t_factor":2},"pid":{"enable":true,"p_factor":50,"i_factor":0.006,"d_factor":10000,"dt":180,"minTemp":5,"maxTemp":80},"sensors":{"outdoor":{"type":1,"gpio":12,"bleAddress":"00:00:00:00:00:00","offset":0},"indoor":{"type":1,"gpio":18,"bleAddress":"00:00:00:00:00:00","offset":0}}}' is not a NumberminTemp":20,"maxTemp":43},"dhw":{"enable":true,"target":45,"minTemp":30,"maxTemp":46},"equitherm":{"enable":false,"n_factor":0.7,"k_factor":3,"t_factor":2},"pid":{"enable":true,"p_factor":50,"i_factor":0.006,"d_factor":10000,"dt":180,"minTemp":5,"maxTemp":80},"sensors":{"outdoor":{"type":1,"gpio":12,"bleAddress":"00:00:00:00:00:00","offset":0},"indoor":{"type":1,"gpio":18,"bleAddress":"00:00:00:00:00:00","offset":0}}}, template: {{ value_json.heating.maxModulation|int(1) }})

Laxilef, 2024-11-17:
Hi,

Remove device from Home Assistant and reboot the ESP.

Pajuris, 2024-11-17:
Thanks, it helped.

closed 2024-11-17 by Pajuris