Laxilef/OTGateway #94: Equitherm function once target temp reached

Daveblanche, 2024-11-18:
Hi there
I’ve looked around and can’t find the answer, yet.
I’m still staying to tune the correct heat curve…my house feel heavy with underfloor heating lag. I think because it has deep screed and insulation below it.
But when I get there, tomorrow, probably, I have a newbie question. 
if set to function with equitherm, N 3.5, K 3.0 and T 2.0, what happens when the target temp is reached?

Does the flow temp reduce to the N value requirement, or below if it overshoots?
Or does the heating switch off until a demand for heat is made again when the temperature drops below target?

I‘m sure this is a stupid question but it kept me awake last night. :)

Laxilef, 2024-11-18:
Hi,

If hysteresis is enabled, the heating will turn off when the indoor temperature is above ``target temperature + hysteresis``.
Equitherm simply calculates the heat carrier temperature.
I think that an N coefficient of 3.5 is a very large value for a well-insulated house with a water-heated floor.

Daveblanche, 2024-11-18:
Hi again,
Thanks for the answer. I appreciate it.
Perhaps my house is „Cardiff“ well-insulated and probably not as good as a new build in Chelyabinsk.
I think the design flow temp is around 40c at -2c outdoor temp.
But I also have a feeling that the underfloor pipe  centres are well above 200mm, though.
But the house is still cheap to heat.

closed 2024-11-18 by Daveblanche

Laxilef, 2024-11-18:
> I think the design flow temp is around 40c at -2c outdoor temp.

This is a fairly low temperature, which indicates good insulation.
I would start with N=0.4, K=5.0, T=2.0 and increase the coefficient if this is not enough.

You can look at the temperature graph [in the calculator](https://github.com/Laxilef/OTGateway/blob/master/assets/equitherm_calc.xlsx) that I published.