Laxilef/OTGateway #95: De Dietrich MPX 20/24 MI COMPACT can't turn heating off

mementoars, 2024-11-20:
Hello everyone,

I have a De Dietrich MPX 20/24 MI COMPACT successfully connected to a D1 mini 32+diyless master opentherm shield.
  Version : 1.4.5
  Apartment: attic
  Indoor sensor: LYWSD03MMC
  External sensor: DS18B20
  Equitherm: enabled (but the coefficients are not adjusted correctly yet).

Boiler shows as a Baxi Luna Duo-Tec
[image]

The controller is connected to home assistant and displays the following:
[image]

Issue No1:
I can't turn the heating off.

[image]

Combinations that didn’t fix the problem:
[image]

issue No2:
The circulator pump doesn’t stop.
Even if I set the desired room temperature to the lowest value (5°C), the circulator pump keeps running and the burner runs periodically to maintain the set temperature.
Is this normal?

Note: when Summer/winter mode is enabled in Opentherm settings the heating and flame turn off)

thank you in advance

Laxilef, 2024-11-20:
Hi,
Set parameter P67=0 in the boiler service menu.

mementoars, 2024-11-26:
That was it!!!
Not all parameters were available (including P67) until I changed P22 from 0 to 22. So I set P67=0 and the problem got solved.
[image]

Thank you so much for the help and the amazing firmware!