Laxilef/OTGateway #96: MQTT seems not to work

Bertus3, 2024-11-23:
Hi, I have gotten this build up and running. Really nice build!
MQTT however does not seem to work. I have tried the latest release (1.4.6), diffrent networks, diffrent MQTT passwords, nothing seems to work yet.

I am running eclipse mosquitto on docker. I am able to publish from other devices and other subnet but the gateway does not seem to be able to connect as seen from the MQTT brokers logs. 

Can anyone help me?

Laxilef, 2024-11-23:
Hi,

Without logs I can't help.
Perhaps you are using a login or password longer than 32 characters?

Bertus3, 2024-11-23:
Here is a few minutes worth of output, set to verbose. Boiler is not connected btw. You can see me change the mqtt server fromt mqtt://.... to just the ip adress, does not make a diffrence. Passwordt is 12 char A-Z a-z 0-9

Couldn't be because it's not connected to a boiler that it is not publishing anything right?

```
[00:04:08][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:04:09][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:04:10][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:04:11][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:04:12][MAIN][VERB] Free heap size: 25072 of 81920 bytes (min: 20120, diff: 0), max free block: 24616 (min: 17568, diff: 0, frag: 1%)
[00:04:12][OT][T[00:02:54][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attemp                                                                                                                                   t:  1   Status: TIMEOUT
[00:02:55][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attemp                                                                                                                                   t:  2   Status: TIMEOUT
[00:02:56][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attemp                                                                                                                                   t:  3   Status: TIMEOUT
[00:02:57][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:02:58][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:02:58][OT][WARN] Failed receive boiler status: TIMEOUT
[00:03:00][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:03:01][MQTT][INFO] Connecting to mqtt://192.168.1.203:1883...
[00:03:01][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:03:02][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:03:03][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:03:05][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:03:05][OT][WARN] Failed receive boiler status: TIMEOUT
[00:03:06][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:03:07][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:03:09][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:03:10][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:03:10][MAIN][VERB] Free heap size: 25520 of 81920 bytes (min: 20376, diff: 0), max free block: 25080 (min: 18680, diff: 0,                                                                                      frag: 1%)
[00:03:11][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:03:11][OT][WARN] Failed receive boiler status: TIMEOUT
[00:03:13][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:03:14][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:03:15][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:03:16][MQTT][INFO] Connecting to mqtt://192.168.1.203:1883...
[00:03:16][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:03:17][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:03:17][OT][WARN] Failed receive boiler status: TIMEOUT
[00:03:19][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:03:20][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:03:21][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:03:22][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:03:23][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:03:23][OT][WARN] Failed receive boiler status: TIMEOUT
[00:03:25][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:03:26][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:03:27][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:03:28][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:03:29][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:03:29][OT][WARN] Failed receive boiler status: TIMEOUT
[00:03:30][MAIN][VERB] Free heap size: 25520 of 81920 bytes (min: 20376, diff: 0), max free block: 25080 (min: 18680, diff: 0,                                                                                      frag: 1%)
[00:03:31][MQTT][INFO] Connecting to mqtt://192.168.1.203:1883...
[... cut ...]
[00:06:04][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:06:05][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:06:06][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:06:07][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:06:09][OT][TRACE] ID:    0   Request: 80001300   Response:        0   Attempt:  5   Status: TIMEOUT
```

Laxilef, 2024-11-23:
No need to use ``mqtt://`` prefix

Laxilef, 2024-11-24:
I don't see any logical reasons.
You need to check the port, login and password. Can you connect to the broker with these credentials via mqtt explorer?

Bertus3, 2024-11-24:
For some reason it started working. Don't know why or how. Thanks anyway!

closed 2024-11-24 by Bertus3