Laxilef/OTGateway #98: Immergas Victrix Zeus 25 enters summer mode

flexmatics, 2024-11-25:
Hi,

I've previously [reported](https://github.com/Laxilef/OTGateway/issues/3#issuecomment-2453361792) support for this boiler. It worked fine for a while, but now noticed that boiler enters summer mode without any changes to settings. Restart boilder/controller didn't help - only disconnect of controller helps.

What can I try to understand where the issue is?

Laxilef, 2024-11-25:
Hi,

What do you mean by summer mode? What is the boiler's behavior?

flexmatics, 2024-11-27:
This is the behavior of boiler when controller is turned on (see video). Perhaps some logs would help to understand what is happening? The strange thing is that this started happening out of the blue without changing anything in settings/setup.

https://github.com/user-attachments/assets/3bca72a6-427c-4a84-91fd-5ed1d525a1c5

Laxilef, 2024-11-27:
Try turning on the "Summer/Winter mode" option.

flexmatics, 2024-11-27:
1. Tried this with on and off - no changes.
2. No, no reaction neither with short press, nor long press.

Perhaps telnet log suggests anything?

> [00:01:57][OT][WARN] Failed receive exhaust temp
[00:01:58][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; fault: 0; diag: 0
[00:01:59][OT][DHW][NOTICE] Received temp: 54.30
[00:01:59][OT][DHW][WARN] Failed receive flow rate
[00:01:59][OT][HEATING][NOTICE] Received temp: 44.60
[00:01:59][OT][HEATING][NOTICE] Received return temp: 47.50
[00:02:00][OT][WARN] Failed receive exhaust temp
[00:02:01][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; fault: 0; diag: 0
[00:02:01][OT][DHW][NOTICE] Received temp: 54.30
[00:02:01][OT][DHW][WARN] Failed receive flow rate
[00:02:01][OT][HEATING][NOTICE] Received temp: 44.60
[00:02:02][OT][HEATING][NOTICE] Received return temp: 47.50
[00:02:02][OT][WARN] Failed receive exhaust temp
[00:02:03][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; fault: 0; diag: 0
[00:02:03][OT][NOTICE] Received min modulation: 0%, max power: 0 kW
[00:02:03][OT][NOTICE] Set max modulation: 100%
[00:02:03][OT][DHW][NOTICE] Received min temp: 10, max temp: 60
[00:02:04][OT][HEATING][NOTICE] Received min temp: 20, max temp: 85
[00:02:04][OT][WARN] Failed receive pressure
[00:02:04][OT][DHW][NOTICE] Received temp: 54.30
[00:02:04][OT][DHW][WARN] Failed receive flow rate
[00:02:05][OT][HEATING][NOTICE] Received temp: 44.50

Laxilef, 2024-11-27:
Hysteresis turns off the heating and this is normal.
Perhaps the boiler interprets the heating being turned off as summer mode. But when the indoor temperature is below the target, the heating should turn on.

What happens if you set the target temperature 2 degrees higher than the indoor temperature? (with equitherm on)

flexmatics, 2024-11-29:
Thanks. I've tested your assumption and it is exactly what happens. If temperature above target, heating is turned back on.

closed 2024-11-29 by flexmatics