Laxilef/OTGateway #103: No MQTT connection after upgrade to 1.5

Pajuris, 2024-12-06:
Strange, but I can not connect to Home assistant Mosquitto broker after update.
After few restarts (HA and OTGateway), opentherm gateway had appeared in HA. But after next OTGateway restart no connection again.

Laxilef, 2024-12-06:
Hi,
Please upload logs

Pajuris, 2024-12-06:
From where?

Pajuris, 2024-12-06:
Solution: need check MQTT enable in Settings. :)
[image]
Sorry...

closed 2024-12-06 by Pajuris

Laxilef, 2024-12-06:
> * The names of some settings have been changed in this version. Not all settings will be restored from backup! Check them after restoration.
> * Sensor settings will not be restored automatically; they must be configured again on the Sensors settings page.
> * Very recommended to delete the device in Home Assistant before upgrading to avoid errors in the HA logs (due to the deletion of some entities)

Pajuris, 2024-12-06:
Thanks, Laxilef...
O have one more question. No switches turn on/off heeating and DHW in Home Assistant after update? Removed or I need to reenable its?

Laxilef, 2024-12-06:
Some entities have been removed or renamed. It is better to remove the device and add it again so that only working ones remain. To control heating and DHW, you must use the "climate" entities.

Pajuris, 2024-12-07:
OK. Thanks, Laxilef.