Laxilef/OTGateway #104: Modulation level and power

Pajuris, 2024-12-07:
I want to talk about the modulation (%) and power (kW).
I think 2 % modulation level when boiler power is 4.95 kW is not logically true when max power can be 24 kW only.
I know that 0 % modulation level is the minimal power of boiler (4.56 kW), but I want to know - this logic is in the boiler or in this software and can it be changed?

Laxilef, 2024-12-07:
4.56 + ((24 - 4.56) * 0.02) = 4.95

But if it is not a condensing boiler, then it cannot have a minimum power of 4.56 kW.

Pajuris, 2024-12-07:
Not that... I want to know why 4.56 kW is 0% moduliation? If 24 kW is 100% modulation, then 4.95 kW should be about 20%..?

Boiler condensig.

Laxilef, 2024-12-07:
> Boiler condensing.

Ok

> I want to know why 4.56 kW is 0% moduliation?

Probably the boiler will not work if 1% gas is supplied. In the specification for boilers there is always a parameter “Minimum thermal power”; it corresponds to 0% modulation.

closed 2024-12-30 by github-actions[bot]