Laxilef/OTGateway #105: Hysteresis 0 - Heating not stopped

Imaginous, 2024-12-09:
My Intergas Xtreme 30 heater slowly goes into a "thermal runaway" because it's running at modulation level 0% but still producing too much heat.

However the thermostat was set to 19.5 degrees and the current roomtemperature is also at 19.5 degrees.
I set the hysteresis degrees to 0, I was expecting that when the desired temperature was reach it would stop heating.
But it doesn't.

My possible solutions:

1. If hysteresis is set to 0, directly stop heating.
2. If the desired temperature is reached but the current temperature is (slowly) "running away" from the setpoint temperature (or more then 1 degree off) then stop heating.

I wonder what your point of view is on this "feature".

Running the latest V1.50 build.

Imaginous, 2024-12-09:
I don't want any hysteresis, set it to 0.
I expect that the OTGateway will stop heating as soon as it hits the set temperature.

This doesn't happen.

Laxilef, 2024-12-09:
You must correctly adjust the coefficients for pid/equitherm (which is used). Hysteresis should not be used instead.

Imaginous, 2024-12-10:
I will do some more tuning.

When does the system decide to stop heating, when the Equitherm or PID setpoint is below the minimum temperature of the heater?

Laxilef, 2024-12-10:
The boiler will automatically switch off the burner when the heat carrier temperature is above the setpoint.

The regulators do not stop the heating, they calculate the temperature of the heat carrier.

DallasJWilm, 2024-12-14:
I just upgraded to 1.5.0, so I could be wrong, but if it is anything like the previous versions, the setpoint is WATER setpoint, NOT room setpoint.  So probably what is happening is the boiler sees the call for heat (b/c Heat = enable), but is trying to limit the water temp (which it can't because in your case air temp = water temp).  For my Rinnai boiler, it turns the burner off and just cycles on to ~0% mod, to off and back and forth when that happens.  Apparently your boiler throttles to minimum modulation but keeps running.

For my case, heating concrete in a garage, I wanted to limit the the cycling.  So I had Home Assistant set water temp to + 30 deg F from room temp, then command flame = on/  Heat = enable for heat, and then command flame = off/ Heat =disable to turn off heat.  This way I do not have constant burner on/off cycling.  This was working, but for some reason lately I am having wi-fi reliability issues, so Home Assistant cannot get its commands to the ESP32 to tell it heating = enable/ heating = disable.

I believe OP's question/issue is related to a **feature request** of mine: Have the ability to tie Heating Enable to the indoor room temperature.  ESP32 thermostat is then a room temperature thermostat.  In my case, water temp would be a constant and heating = enable is based on room temperature.  This way, I would not be at the mercy of the wi-fi connection with home assistant to control the room temperature.

Laxilef, 2024-12-15:
@Imaginous, when using Equitherm + PID, Equitherm simply calculates the heat carrier temperature based on the outdoor temperature. PID additionally adjusts the heat carrier temperature until the indoor temperature is equal to the setpoint. The influence of the PID on the heat carrier temperature is limited by the minimum and maximum PID temperature settings.
Accordingly, when the indoor temperature is higher than the setpoint, the PID will decrease the temperature of the heat carrier. And increase it when the indoor temperature is lower than the setpoint.

Laxilef, 2024-12-15:
@DallasJWilm, enable and configure Equitherm or PID to regulate indoor temperature.

DallasJWilm, 2024-12-15:
Hi Laxilef,

Thanks, I do realize that equitherm and pid will regulate temperature.  I’d just like the ability to have heat enable be able to be set true/false based on indoor temperature. I think this may help the OP also. In any case, this is great software that gave me great insights to my boiler and I appreciate your hard work!

Laxilef, 2024-12-15:
Hi @DallasJWilm,
Well, you can create automation in your home assistant that will turn on/off the heating depending on the indoor temperature

DallasJWilm, 2024-12-15:
Thanks, I did do that.  It was working well until a couple weeks ago when wi-fi started dropping out and then HA couldn't reach it.  Signal is only about 55% or so, so that could be the problem.  I also tried upgrading to 1.5 to see if that has any effect.

Laxilef, 2024-12-15:
Is this a normal signal level, maybe the problem is something else? Try using static IP instead of DHCP

DallasJWilm, 2024-12-16:
Yes, that is about normal.  I already had a DHCP reservation, but I will try static.

closed 2025-01-05 by github-actions[bot]