Laxilef/OTGateway #106: Modulation level doesnt work after upgrade

shaggee, 2024-12-09:
Today I upgraded D1 Mini ESP 32 from 1.4.0 to 1.5.0 and modulation level stays always 0% even boiler flame is on ant its heating.
Any advice to fix it?

[image]

Laxilef, 2024-12-09:
Hi,

Pls upload logs

shaggee, 2024-12-10:
> Hi,
> 
> Pls upload logs

[logs.txt](https://github.com/user-attachments/files/18074718/logs.txt)

I found more missing diagnostic info which were visible in 1.4.0:
Boiler DHW max temp
Boiler DHW min temp
Boiler heating max te
Boiler heating min te
Diagnostic code
Modulation level

Laxilef, 2024-12-10:
> Boiler DHW max temp
> Boiler DHW min temp
> Boiler heating max te
> Boiler heating min te

```
[09:07:33][OT][DHW][NOTICE] Received min temp: 35 (converted: 35), max temp: 65 (converted: 65)
[09:07:33][OT][HEATING][NOTICE] Received min temp: 25 (converted: 25), max temp: 48 (converted: 48)
```

> Modulation level

```
[09:08:10][OT][NOTICE] Received modulation level: 38%, power: 14.46 of 25.00 kW (min: 8.00 kW)
```

> Diagnostic code

Not polled if fault/diag state is inactive

---

I see no problems in the logs.

closed 2024-12-10 by shaggee