Laxilef/OTGateway #107: TEMP1 NOT WORK

freeupmao, 2024-12-10:
TEMP1 NOT WORK，TEMP1 is not functioning properly, while TEMP2 is running normally. Using a multimeter to check the voltage, TEMP2 shows 3.3V and TEMP1 shows 0V.

My chip is WeMos Lua D1.

Laxilef, 2024-12-10:
Hi,

What adapter/board is used? Firmware version?

freeupmao, 2024-12-10:
https://oshwlab.com/lazydev/otgateway

freeupmao, 2024-12-10:
Version V.1.4.1 of this project's

> Hi,
> 
> What adapter/board is used? Firmware version?

The firmware was upgraded to version 1.5.0 today, but TEMP1 is still not working."

Laxilef, 2024-12-10:
Check R2 on the board. This is a pull-up resistor.

upais, 2024-12-17:
My NTC sensor doesn't work on 1.5.0. ESP WROOM-32 board, connected to T2 (GPIO 26).
There were no problems on 1.4.6.

The logs only show this:
[00:01:48][SENSORS][NTC][WARN] GPIO 26, sensor #0 'Outdoor temp', voltage is out of threshold: 0.000

Laxilef, 2024-12-17:
> My NTC sensor doesn't work on 1.5.0. ESP WROOM-32 board, connected to T2 (GPIO 26).
> There were no problems on 1.4.6.

NTC sensors were not supported on 1.4.6

upais, 2024-12-17:
Sorry, thought it was them. Chose DALLAS and the problem was solved.

closed 2025-01-06 by github-actions[bot]