Laxilef/OTGateway #110: Heating not stopping

cybuletz, 2024-12-18:
Hi, 

I would like to understand what is the logic behind starting and stopping heating. 
Sometimes, the climate entities stay on heating for up to 10 minutes before the heating starts and also the opposite, sometimes, the heating stops late even though the heating temperature is above the setpoint (see picture below).

[image]

Laxilef, 2024-12-19:
Hi,

This is normal. The burner turns on when the heat carrier temperature is below setpoint-N. And it turns off when the heat carrier temperature is below setpoint+N.

N - boiler hysteresis, sometimes can be changed in the boiler service menu.

Laxilef, 2024-12-19:
https://github.com/Laxilef/OTGateway/wiki#fit-coefficients

Laxilef, 2024-12-19:
Look at the settings in the boiler service menu

closed 2025-01-08 by github-actions[bot]