Laxilef/OTGateway #111: Failed receive boiler status: TIMEOUT

kuqqum, 2024-12-21:
Hi,

I have connected the Diylless board and configured as below. Everything looks fine accept the OpenTherm connection. My boiler is ATAG and can work with opentherm communications.
Is there a way to check this configuration.

[image]

[image]

Laxilef, 2024-12-22:
Hi,

1. Check GPIO
2. Also recommend studying the documentation for the boiler; may need to change some settings in the service menu of the boiler for opentherm devices to work.

Without detailed information I can't help.

kuqqum, 2024-12-22:
Hi,

1. GPIO settings are OK.

2. The ATAG boiler is connected (wired) and working over opentherm with "ATAG One"  thermostat. Ther's nothing I found in the documentation to change in the settings that can block communication.
3. The boiler is working properly after removing the thermostat and connecting the openTherm gateway without wornings or errors.

In the logs from MQTT there is a strange ip-address for homassistant after restart. This ip-address is not insite our network.

I hope you can help.

[image]

Laxilef, 2024-12-22:
Write the exact name of the adapter and ESP board.

> In the logs from MQTT there is a strange ip-address for homassistant after restart. This ip-address is not insite our network.

This is a docker network.

kuqqum, 2024-12-22:
The ESP board is ESP32_Devkitc_V4 with ESP32-VROOM-32D
And OPENTHERM MASTER SHIELD DIYLess.com.

Laxilef, 2024-12-22:
Can you upload a photo?
This shield is mainly used with Wemos type boards.

kuqqum, 2024-12-23:
Hi,
It's my intention to integrate ESP32_Devkitc_V4 with OT gateway into my floor heating system but that wasn't a good idea.  Now I flashed OT gateway into d1_mini32 and it's, standalone, working fine.

DHW in my boiler is only tapwater heating up for direct usage. Is it possible to manage this with OT gateway also?

Laxilef, 2024-12-24:
Any esp32 will work. The problem was probably in the wrong gpio, that's why I asked for a photo.

> DHW in my boiler is only tapwater heating up for direct usage. Is it possible to manage this with OT gateway also?

Of course

kuqqum, 2024-12-24:
Ok. Thanks. Then there is something missing. When I tap water, the boiler doesn’t start heating up. A three way valve has to change floating direction, but I don’t hear changing it. What can I do to find out what is going wrong. Joop

kuqqum, 2024-12-24:
I checked again as you can see below.
DHW is not active after opening a tap.

Dashboard shows this:
[image]

OpenTherm settings are this:

[image]

kuqqum, 2024-12-24:
> DHW in my boiler is only tapwater heating up for direct usage. Is it possible to manage this with OT gateway also?

> Of course

Then I don't understand how my system ATAG E325EC knows there's need for hot tapwater. OT is not reporting a DHW flow rate.

kuqqum, 2024-12-24:
I asked chatGPT this question voor a E325EC, this is the answer:
To determine whether there is heat demand for domestic hot water or central heating:

Monitor the status of the boiler via the OpenTherm gateway or other reading device.
Check the status byte (message ID 0):
If Bit 0 (CH enable) is active: there is demand for CH.
If Bit 1 (DHW enable) is active: there is demand for hot water.
Check the current flow temperature and flame status as an additional check.

Laxilef, 2024-12-25:
Try OT option ``Duplicate DHW to CH2`` or/and ``DHW blocking``.

kuqqum, 2024-12-25:
All tree possibilities are with negative results.

Laxilef, 2024-12-26:
Well, I have no other suggestions 🤷‍♂️

kuqqum, 2024-12-26:
OK.
I will try to find out if there is any usage full information coming from the boiler after opening a HW tap.
It's absolutely unclear for me how this boiler detects that there is need for CV or DHW.

Laxilef, 2024-12-26:
There may be a sensor in the boiler that simply tells whether there is flow or not.

kuqqum, 2024-12-26:
In that case will always work independent from thermostat. But with Atag One thermostat DHW works. With OT thermostat not.
So there must be a communication between the Atag One thermostat and the boiler to allow DHW heating.

Is it possible that not all OT commands are visible in the terminal list?

[00:03:53][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:03:54][OT][NOTICE] Received pressure: 1.74 (converted: 1.74)
[00:03:54][SENSORS][NOTICE] #8 'Pressure' new value 0: 1.74, compensated: 1.74, raw: 1.74
[00:03:55][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
[00:03:55][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:03:55][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:03:55][OT][DHW][NOTICE] Received temp: 13.88 (converted: 13.88)
[00:03:55][SENSORS][NOTICE] #5 'DHW temp' new value 0: 13.88, compensated: 13.88, raw: 13.88
[00:03:56][OT][HEATING][NOTICE] Received temp: 30.00
[00:03:56][SENSORS][NOTICE] #2 'Heating temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:03:56][OT][HEATING][NOTICE] Received return temp: 30.00 (converted: 30.00)
[00:03:56][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:03:57][OT][NOTICE] Received pressure: 1.74 (converted: 1.74)
[00:03:57][SENSORS][NOTICE] #8 'Pressure' new value 0: 1.74, compensated: 1.74, raw: 1.74
[00:03:58][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
[00:03:58][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:03:58][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:03:58][OT][DHW][NOTICE] Received temp: 13.62 (converted: 13.62)
[00:03:58][SENSORS][NOTICE] #5 'DHW temp' new value 0: 13.62, compensated: 13.62, raw: 13.62
[00:03:59][OT][HEATING][NOTICE] Received temp: 29.88
[00:03:59][SENSORS][NOTICE] #2 'Heating temp' new value 0: 29.88, compensated: 29.88, raw: 29.88
[00:03:59][OT][HEATING][NOTICE] Received return temp: 30.00 (converted: 30.00)
[00:03:59][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:03:59][OT][NOTICE] Received pressure: 1.74 (converted: 1.74)
[00:03:59][SENSORS][NOTICE] #8 'Pressure' new value 0: 1.74, compensated: 1.74, raw: 1.74
[00:04:00][SENSORS][NOTICE] #4 'Heating setpoint temp' new value 0: 32.14, compensated: 32.14, raw: 32.14
[00:04:01][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
[00:04:01][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:04:01][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:04:01][OT][DHW][NOTICE] Received temp: 13.50 (converted: 13.50)
[00:04:01][SENSORS][NOTICE] #5 'DHW temp' new value 0: 13.50, compensated: 13.50, raw: 13.50
[00:04:01][OT][HEATING][NOTICE] Received temp: 29.75
[00:04:01][SENSORS][NOTICE] #2 'Heating temp' new value 0: 29.75, compensated: 29.75, raw: 29.75
[00:04:02][OT][HEATING][NOTICE] Received return temp: 30.00 (converted: 30.00)
[00:04:02][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:04:02][OT][NOTICE] Received pressure: 1.74 (converted: 1.74)
[00:04:02][SENSORS][NOTICE] #8 'Pressure' new value 0: 1.74, compensated: 1.74, raw: 1.74
[00:04:03][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
[00:04:03][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:04:03][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:04:04][OT][DHW][NOTICE] Received temp: 13.00 (converted: 13.00)
[00:04:04][SENSORS][NOTICE] #5 'DHW temp' new value 0: 13.00, compensated: 13.00, raw: 13.00
[00:04:04][OT][HEATING][NOTICE] Received temp: 29.62
[00:04:04][SENSORS][NOTICE] #2 'Heating temp' new value 0: 29.62, compensated: 29.62, raw: 29.62
[00:04:05][OT][HEATING][NOTICE] Received return temp: 30.00 (converted: 30.00)
[00:04:05][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:04:05][OT][NOTICE] Received pressure: 1.75 (converted: 1.75)
[00:04:05][SENSORS][NOTICE] #8 'Pressure' new value 0: 1.75, compensated: 1.75, raw: 1.75
[00:04:06][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
[00:04:06][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:04:06][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:04:06][OT][DHW][NOTICE] Received temp: 13.62 (converted: 13.62)
[00:04:06][SENSORS][NOTICE] #5 'DHW temp' new value 0: 13.62, compensated: 13.62, raw: 13.62
[00:04:07][OT][HEATING][NOTICE] Received temp: 29.50
[00:04:07][SENSORS][NOTICE] #2 'Heating temp' new value 0: 29.50, compensated: 29.50, raw: 29.50
[00:04:07][OT][HEATING][NOTICE] Received return temp: 30.00 (converted: 30.00)
[00:04:07][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:04:08][OT][NOTICE] Received pressure: 1.76 (converted: 1.76)
[00:04:08][SENSORS][NOTICE] #8 'Pressure' new value 0: 1.76, compensated: 1.76, raw: 1.76
[00:04:09][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
[... cut ...]
[00:04:36][SENSORS][NOTICE] #5 'DHW temp' new value 0: 20.62, compensated: 20.62, raw: 20.62
[00:04:36][OT][HEATING][NOTICE] Received temp: 28.88
[00:04:36][SENSORS][NOTICE] #2 'Heating temp' new value 0: 28.88, compensated: 28.88, raw: 28.88
[00:04:37][OT][HEATING][NOTICE] Received return temp: 29.25 (converted: 29.25)
[00:04:37][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 29.25, compensated: 29.25, raw: 29.25
[00:04:37][OT][NOTICE] Received pressure: 1.44 (converted: 1.44)
[00:04:37][SENSORS][NOTICE] #8 'Pressure' new value 0: 1.44, compensated: 1.44, raw: 1.44

Laxilef, 2024-12-27:
Pls try with Master member ID = 48

Laxilef, 2024-12-27:
> Is it possible that not all OT commands are visible in the terminal list?

Yes
Change log level to VERBOSE to display all messages

kuqqum, 2024-12-27:
I switch off the flowrate in the acceptation there isn't one, but I studied te electrical drawing and find out that there is a flow rate sensor present in the system, OT is not returning a value.
`[03:25:42][OT][VERB] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID`
`[03:25:42][OT][DHW][WARN] Failed receive flow rate`

kuqqum, 2024-12-27:
> 
[image]

Laxilef, 2024-12-27:
The boiler may not report the flow rate, but this does not mean that the DHW will not work.
The firmware does not control the operation of the burner, we simply tell the boiler what temperature it needs to maintain.

Laxilef, 2024-12-27:
I don't know why your DHW doesn't work. For other boilers this code works correctly. Try to set the Master member ID = 48

kuqqum, 2024-12-27:
WOW, you dit a great job. Flow rate is still not working but wat the heck. We don't have to shoer anymore with cold water...
Wat is the meaning of member ID = 48?
Thank you so much...

Laxilef, 2024-12-27:
> Wat is the meaning of member ID = 48?

This is the member id of the original ATAG thermostat.
Probably, there is a vendor lock in your boiler software. Thus, they prohibit the use of non-original ATAG thermostats.

kuqqum, 2024-12-30:
Oké. I'l do that. Gif me sum time. I'm very happy now.
Thanks again.

closed 2025-01-19 by github-actions[bot]