Laxilef/OTGateway #112: Добавить округление значений в сенсоры

upais, 2024-12-24:
Приветствую.
Переработка настройки сенсоров отличная и удобная. Предлагаю добавить еще и округление значений после запятой. Значение той же температуры в виде 0.13 градусов не особо имеет смысл в такой точности, особенно с учетом погрешности измерений сами датчиков.

Laxilef, 2024-12-24:
Здравствуйте.
Значения округляются до 2 знаков после запятой, потому что просили в чате для сбора статистики. В home assistant можете выбрать округление до целого числа в настройках сущности.

closed 2024-12-24 by Laxilef

upais, 2024-12-24:
1) Кому нужна статистика могли бы округлять до нужного им.
2) Home Assistant не обязательное ПО и не единственный УД.
3) В Home Assistant округление не работает в кастомных касточках, если в них нет такой поддержки. Либо шаблоны.
Тогда уж и коррекцию с фильтрацией можно было на HA переложить.

Ладно, дело автора.

Laxilef, 2024-12-25:
> Кому нужна статистика могли бы округлять до нужного им.

Если мы округлим в прошивке, то потом разокруглить значение не получится)

> Home Assistant не обязательное ПО и не единственный УД.

Это верно, поэтому сделано универсально. Если в другой системе УД есть возможность округления, то там это тоже можно сделать.

> В Home Assistant округление не работает в кастомных касточках, если в них нет такой поддержки.

В т.ч. в кастомных карточках же просто отображается значение сущности, если в настройках сущности поменять округление, то там оно не применяется?
[image]

upais, 2024-12-27:
> В т.ч. в кастомных карточках же просто отображается значение сущности, если в настройках сущности поменять округление, то там оно не применяется?

Как понимаю, HA не округляет значение до формирования самого сенсора, а делает это уже после и отображает в карточках. Но значение самого сенсора не меняется.
Вот пример. В панели разработчика тоже без округления.
[image]

Да, я понимаю, что сама доработка не имеет большой ценности, но она логично вписывается и если не трудозатратна, то почему бы и нет.