Laxilef/OTGateway #113: Channel 1 and 2 independent activation based on indoor thermometer additional entity

Daveblanche, 2024-12-26:
Hi there,
With the changes you made recently to the firmware, does it look like it might be possible to make it so that channel 1 and channel 2 can call for heat independently based on the same equitherm settings?
My reason for asking:
System is ufh on channel 1…very slow and stable. Very efficient. Very smooth.
Channel 2 is a radiator based circuit which functions really when mirrored to channel 1, except for 1 room which has the garage underneath it so cools down much more quickly.

What this sometimes causes is the heating downstairs to switch off, if we are cooking or partying/creating heat, and that one room gets a lot colder.
I can cheat by just setting the basic equitherm curve to the heat loss for the home, with no k and T values, but then I am missing out on all the shift and other sexy settings the laxilef system provides.

all radiators have tado thermostats on them acting as temperature limiters only.

Laxilef, 2024-12-26:
Hi @Daveblanche
Does your boiler support two separate heating circuits?

Daveblanche, 2024-12-26:
Thanks for the quick response!

Yes, I think so. The system runs two circuits separately. Viessmann B1hc. So the radiator circuit (tado) can call for heat and the ufh (Heatmiser) can do so too…when not using the open therm system.

Laxilef, 2024-12-26:
Can you take a photo of the pipes connecting to the boiler? I knew that the protocol supports a second channel (circuit), but I have not seen such a boiler. Usually only DHW and one heating circuit.

Daveblanche, 2024-12-26:
Ok. Will do. It is an s-plan, I think.

I’ve drawn onto the pictures for a bit of clarity, I hope. Just remember that I do bones and biomechanics. 

Viessmann B1hc
Top right is the motorized valve that controls just the underfloor heating. 
the main pipe coming out the boiler on the left side is the one that runs the ufh.
The one bottom right horizontal, is the dhw priority motorised valve which also (I think) connects with the demand box for dhw, and the internal selector valve on the boiler.

So if the boiler is running just for the radiators, the top right valve is open.
If no heat is required for the ufh, then the separate pump for the ufh does not turn on, so no flow to ufh. (I think)

if ufh only is needed then the pump for ufh turns on and the channel 2 (radiator valve closes).

if both are needed then the ufh pump runs and the channel 2 valve opens.  Boiler main pump is running in all cases above.
the ufh pump only turns on when needed.

I also have the install manual for the boiler.
Happy to play sacrificial lamb here as a I think you should be getting these built and sold for income.
Also happy to help with any literature for users.

[image]
[image]
[image]

Laxilef, 2024-12-26:
Wow thanks
I need time to research this 😅

Laxilef, 2025-01-08:
Hi @Daveblanche

So far I don’t have any ideas on how this could be easily implemented. If we add another "thermostat" to control the second channel, we will have to add new indoor temperature sensors, Equitherm and PID instances for calculating the heat carrier temperature.
Implementing this feature requires a huge refactoring of the code, and unfortunately I don’t have free time for that right now.

If you have experience in programming, you can try to implement this feature and create a PR :) Or, if you have ideas on how this can be implemented, pls write them.

Daveblanche, 2025-01-11:
Hiya. I will have a look at this in a couple of weeks. I’ve never done any coding, but will have a go!
Will buy another Diyless thermostat and OpenTherm board.
I might need some pointers though, if you can sense check me.

closed 2025-01-11 by Daveblanche