Laxilef/OTGateway #114: Strange problem

salvo76, 2024-12-27:
Hi, I installed the card on a Ferroli Bluehelix 24, I configured everything so that it can be seen by home assistant, and so far everything is ok, I can read all the water flow and return values, the burner, if I use hot water (DHW) I have the correct values ​​etc. etc. But on the heating part there is something that I can't get to work: all the values ​​seem correct, the opentherm reads the boiler but the heating doesn't start at all, I tried both equiterm, pid, or native of the boiler, but nothing, I attach a screen shot, could someone suggest something?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Laxilef, 2024-12-27:
Hi,
Show your OT settings

Laxilef, 2024-12-27:
Set:

✅ DHW present
❌ Summer/winter mode
✅ Heating CH2 always enabled
❌ Duplicate heating CH2 to CH2
❌ Duplicate DHW to CH2
❌ DHW blocking
❌ Sync modulation with heating
✅ Get min/max temp from boiler
❌ Native heating control

Restart the ESP, set the heating to 50 degrees and wait 5-10 minutes

Laxilef, 2024-12-27:
[image]
Have you removed the jumper from contacts 1-2?

salvo76, 2024-12-27:
Thanks, now the heating has started, already for 10 minutes but as a reference it has the temperature of the water in the radiators (50 C°), how can I manage it with the internal and external thermostats? The jumper is certainly removed

Laxilef, 2024-12-27:
Enable and configure Equitherm or/and PID.

salvo76, 2024-12-27:
Ok, thanks, I thought so too. One last thing, regarding the modulation percentage and the power in Kw, they are always at 0, is there a way to display them? It would be useful for me to have this data in heating.

Laxilef, 2024-12-27:
If the boiler does not report the modulation level, it is impossible to calculate the power in kW

salvo76, 2024-12-27:
On the DHW part, I can see the modulation percentage and it works, it oscillates according to the water required, but when I turn on the heating, it starts at about 37% and then drops to 0% after about 30 seconds, and it doesn't move anymore.

Laxilef, 2024-12-27:
It's normal. 0% modulation is the minimum power.
Configure minimum and maximum power in OT settings to display the current power in kW.

closed 2025-01-16 by github-actions[bot]