Laxilef/OTGateway #115: 1.5.0 Stability issues when boiler is not supporting enabled sensors

SanFable, 2024-12-28:
Hello,

I’ve been using OTGateway with my Beretta Ciao Green 25 C.S.I. boiler for about a year. It works fine with smart TRVs and Home Assistant, although the functionality is somewhat limited.

Recently, I updated to version 1.5.0 and switched to an ESP32-S3 (I also tried the ESP32-C3 with the same results). Previously, I was using version 1.4.5 with an ESP8266.

**The Issue**
After the update, the system became unstable:

- After about a minute, the OpenTherm Gateway status shows "problematic," and everything becomes unavailable.

- The ESP32-S3 seems to struggle:

- The web interface is very slow and unresponsive.
- Logs over Telnet are delayed and limited.
- With the ESP32-C3, the system was completely unreachable.

**What I Found**
I think the issue is caused by new sensors/IDs, such as:

- Return temperature
- Flow rate
- Exhaust temperature
- Pressure
- Minimum modulation
- Maximum power

In the logs, I saw warnings like:
`[WARN] Failed to receive ...`

It looks like the boiler doesn’t support these IDs, and they might be overloading the OpenTherm communication, causing it to crash.

**What Worked**
I turned off these new sensors and reconnected the OTGateway to the boiler. Since then, it’s been running stable for over 30 minutes, and the OpenTherm Gateway status is "OK."

Attaching [logs_1.txt](https://github.com/user-attachments/files/18266836/logs_1.txt) from a setup where I didn’t disable all the mentioned sensors (minimum modulation and maximum power were still enabled).
In the end, the ESP32 crashed, and the Telnet connection was lost.

I just realized that even after turning off the mentioned sensors in my successful run, I’m still seeing invalid request IDs 15 and 14 (minimum modulation, maximum power, and maximum modulation).

Attaching [logs_2.txt](https://github.com/user-attachments/files/18266837/logs_2.txt) from over 30 minutes of stable operation. However, there are still warnings in the logs from sensors that should be disabled. Could it be that something is overlapping when these sensors are enabled?

PS is there anything that I could do to improve support for my boiler?

tincanpete, 2024-12-29:
I believe I'm having the same issue. The symptoms are the same, and I have also recently upgraded to 1.5.0 but have not been able to get it running in any stable or reliable way. 

I think the issue might have something to do with MQTT as I get better results with it turned OFF but have not absolutely verified this yet.

Daveblanche, 2024-12-29:
Hi both,
Quick question, but did you delete the HA instance of OpenTherm gateway on  mqtt before updating the firmware or chipset. Yuri warned of potential errors, if this was not done.

tincanpete, 2024-12-29:
Thanks @Daveblanche for the tip, but yes I definitely did do that as recommended. I think the issue is on the OT Gateway end not the HA end though. I'm going to continue to try to figure this out and will post more here as I make progress.

SanFable, 2024-12-29:
@Daveblanche
Before setup I have went to HA settings ->devices and services->mqtt->devices, selected opentherm and removed it. Then after installing new one I removed and added new cards in dashboard.

Regarding my stability, I had few reconnections yesterday, but at 23:00 I disabled logging (serial and telnet) and it failed only once a whole day for 40 seconds.

Laxilef, 2024-12-31:
Anyone who has problems with losing connection, test this build.
[1.5.1-testing.zip](https://github.com/user-attachments/files/18281198/1.5.1-testing.zip)

**And happy holidays!**

tincanpete, 2024-12-31:
Thank you @Laxilef I will give it a try!

SanFable, 2024-12-31:
@Laxilef @tincanpete 

I have tested 1.5.1,
I have setup it like normal, kept telnet logs enabled, for 20mins I stayed with default settings and warns because boiler low support (just filled the mqtt section) and everything was stable. After that I have disabled sensors that are not existing in my boiler (thats shame its lacking some useful sensors...)

no reconnection since 2 hour, it works now. Page is responsive. No complaints. This state was not achievable on 1.5.0, Good job :)

I'm interested about @tincanpete feedback.

tincanpete, 2024-12-31:
Hello, I have also just tested 1.5.1 and while I thought there was some initial success, its seems no. Having MQTT enabled definitely makes things significantly worse.

I have attached a file showing repeated 'ping' to the board. When it's working well, I always see about 10-50ms. However as you can see it quickly becomes unstable and unresponsive; and will eventually sort-of come back to life but it seems quite random. 

During the time when ping is very slow or dropped, the UI is also unresponsive, and if connected, the MQTT server will report the device is off-line. Sometimes my boiler will also report an OpenTherm communication error on its display. 

When the problem goes away, everything goes back to normal and works OK, but often not long enough to be useful. 

I'm using S2-mini board, and this ping trace was done with the gateway serial port, telnet, and logging all turned OFF to try and make sure that high level logging wasn't causing high load to be part of the problem. 

[ping example.txt](https://github.com/user-attachments/files/18282836/ping.example.txt)

Thanks for your help!

SanFable, 2024-12-31:
@tincanpete 
try following:
download current settings backup
reflash whole esp32 using PC (flash factory image and filesystem like new one)
when connecting to esp32 for the first time restore settings. 
connect esp32 to the boiler.

In my case esp32-s3 seems fine, but when I changed to weaker esp-c3 it was nightmare, just not working. I guess I would have similar results as you.

If you still have problems maybe try disabling sensors that are not available on your boiler (to minimize warnings in logs)

I belive my [pings.txt](https://github.com/user-attachments/files/18282880/pings.txt) are fine, its wifi with signal 68-76% reported on OTG page.

tincanpete, 2024-12-31:
@SanFable I will try that, I did not use the "factory" image, just the normal one and upgraded via the UI. 
I will report back my progress!
Thanks

tincanpete, 2025-01-01:
Still no joy unfortunately even after fully erasing and re-flashing the S2 and using the factory bin. 

I have attached a larger log but just look at this extract from the end:

[00:02:55][SENSORS][NOTICE] #5 'DHW temp' new value 0: 11.20, compensated: 11.20, raw: 11.20
[00:02:56][OT][DHW][NOTICE] Received flow rate: 0.00 (converted: 0.00)
[00:02:56][SENSORS][NOTICE] #6 'DHW flow rate' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:02:56][OT][HEATING][NOTICE] Received temp: 10.00
[00:02:56][SENSORS][NOTICE] #2 'Heating temp' new value 0: 10.00, compensated: 10.00, raw: 10.00
[00:02:57][OT][HEATING][NOTICE] Received return temp: 10.60 (converted: 10.60)
[00:02:57][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 10.60, compensated: 10.60, raw: 10.60
[00:02:58][OT][NOTICE] Received exhaust temp: 10.00 (converted: 10.00)
[00:02:58][SENSORS][NOTICE] #7 'Exhaust temp' new value 0: 10.00, compensated: 10.00, raw: 10.00
[00:02:58][OT][NOTICE] Received pressure: 0.80 (converted: 0.80)
[00:02:58][SENSORS][NOTICE] #8 'Pressure' new value 0: 0.80, compensated: 0.80, raw: 0.80
[00:02:59][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
[00:02:59][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:02:59][SENSORS] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:03:11][SENSORS][NOTICE] #4 'Heating setpoint temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:03:11][OT][DHW][NOTICE] Received temp: 11.00 (converted: 11.00)
[00:03:18][SENSORS][NOTICE] #5 'DHW temp' new value 0: 11.00, compensated: 11.00, raw: 11.00
[00:03:28][SENSORS][NOTICE] #4 'Heating setpoint temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:03:37][OT][DHW][NOTICE] Received flow rate: 0.00 (converted: 0.00)
[00:03:38][SENSORS][NOTICE] #6 'DHW flow rate' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:03:44][OT][HEATING][NOTICE] Received temp: 10.20
[00:03:53][SENSORS][NOTICE] #2 'Heating temp' new value 0: 10.20, compensated: 10.20, raw: 10.20
[00:03:55][SENSORS][NOTICE] #4 'Heating setpoint temp' new value 0: 30.00, compensated: 30.00, raw: 30.00
[00:03:55][OT][HEATING][NOTICE] Received return temp: 10.80 (converted: 10.80)
Connection closed by foreign host.

Note the time stamps from 2:59 onwards, there's a big delay between each one. 

At the same time, the pings to the board are looking like this screenshot:

[image]

Full log file below:
[1.5.1 log.txt](https://github.com/user-attachments/files/18285697/1.5.1.log.txt)

All of this was with MQTT turned OFF by the way, which I thought would be better, but did not help actually. 

thanks!
Pete

tincanpete, 2025-01-01:
To add to my previous comment, I did try disabling the "power" sensor which my boiler does not support, but it didn't seem to make any difference.

SanFable, 2025-01-01:
From my side, I had 2 reconnections every 24h using esp32s3. I assume esp32 s3 is powerful enough to handle something bad thats going on in the background. C3 was not usable.

Laxilef, 2025-01-01:
Guys, I can assume that the problem may be in the router. Let's check it: check how the web interface works when connected to the ESP access point, i.e. when the network is not yet configured on the ESP.

SanFable, 2025-01-01:
I'm using ubiquiti U7 pro. Everything was fine on previous versions (1.4.5 in my case with mini D1).

When first time configuring connected to ESP access point web interface is blazing fast. I will try to differ if it slows down after connecting to boiler or not.

UniFi controller doesn't show me problems, it says wifi experience excellent (95%+), some spikes to good (88%)

Laxilef, 2025-01-01:
> Everything was fine on previous versions (1.4.5 in my case with mini D1).

Test 1.5.0 on your D1 mini. Now you are testing on ESP32, these are different boards and there is a different SDK.

tincanpete, 2025-01-01:
Interesting idea, I had considered it might be a wifi/router issue, however the network is stable and has been for a long time, running Home Assistant and many other ESP-based devices (Shelly Relays and similar) without a problem. Do you think there's a chance my Wemos S2-mini board just "doesn't like" the wifi network? The problem was present, although not as severe, with software 1.4.5.

Daveblanche, 2025-01-01:
I’m on UniFi, too. Check your retry rates on the front page of UniFi network app. I had an issue a few weeks ago, with high retries, and it was down to channel choice/availability. 

My network is incredibly stable, too.

Laxilef, 2025-01-01:
Something similar, and there was UniFi there too: https://community.home-assistant.io/t/opentherm-gateway-thermostat-with-full-integration-for-home-assistant/617228/128

SanFable, 2025-01-01:
I just have only one U7 Pro, no meshing.
[image]
TX retries, I would say opentherm is in the very middle of the devices. I just have noisy 2.4ghz. I'm on channel 1 20mhz which unifi auto choosen. I will experiment with others and check TX rate.

I will try d1 mini tomorrow.

Laxilef, 2025-01-02:
ESP32 C3 connected to mikrotik, OT not connected

[esp c3.webm](https://github.com/user-attachments/assets/1515f37c-fdbd-4ea9-acfe-92ee6973ef0e)

tincanpete, 2025-01-04:
I've not had a chance to try with Wireless AP only as the hardware is running in a shop we own and I've not been there for a couple of days. However, remotely monitoring it I have just noticed the "Uptime" on the UI homepage has been reset and the "Last Reset Reason" is showing "Reset due to other watchdogs". Does this offer any clues to you?

Laxilef, 2025-01-04:
> Does this offer any clues to you?

[image]

tincanpete, 2025-01-04:
No, "save debug data" just gives me this:
```

{
  "build": {
    "version": "1.5.1-testing",
    "date": "Dec 31 2024 02:02:27",
    "env": "s2_mini",
    "core": "3.1.0",
    "sdk": "v5.3.2-174-g083aad99cf-dirty"
  },
  "heap": {
    "total": 188452,
    "free": 55288,
    "minFree": 48052,
    "maxFreeBlock": 30708,
    "minMaxFreeBlock": 25588
  },
  "chip": {
    "model": "ESP32-S2",
    "rev": 100,
    "cores": 1,
    "freq": 240
  },
  "flash": {
    "size": 4194304,
    "realSize": 4194304
  },
  "crash": {
    "reason": "Reset due to other watchdogs",
    "core": 0,
    "heap": 58416,
    "uptime": 680780475
  }
}

```

Under what circumstances will Watchdog cause a reboot?

Laxilef, 2025-01-04:
Hmm, strange, there is no backtrace in the debug data. Without a backtrace it is impossible to find out the reason.
There may be many reasons, sometimes it is related to poor power supply of the ESP.

Symon84, 2025-01-05:
Hi, I have a similar issue with system instability (frequent disconnections).
The system seems to get saturated.
This happens exactly after modifying the values related to Emergency mode.

By default, I have these parameters:

    Target temperature: 40
    Threshold time: 120

My system is configured with a minimum flow temperature of 50 degrees.
If I change the target temperature to 50 degrees in Emergency mode and set the threshold to 120, the system stops working, becomes unstable, connects and disconnects continuously, and constantly activates Emergency mode.

I’ve tried this three times (always with firmware 1.5.0), and the issue has replicated every time.
The only way to restore the system is to erase the firmware and flash it again.

If I leave the Emergency mode values unchanged (40°C and 120 seconds), everything works correctly.

I hope this can help.
Thank you so much for the amazing work!

Symon84, 2025-01-05:
Never mind… the problem has now reappeared even without modifying the parameters.
I’ll try replacing the ESP8266 (D1 Mini) to see if it’s a hardware issue. I’ll keep you updated.
For the record, it worked perfectly for four consecutive days.

The disconnection issue occurs even if the device is not connected to the boiler.

Laxilef, 2025-01-05:
If your ESP is powered via USB, try replacing the power supply with a different one.

Symon84, 2025-01-07:
> If your ESP is powered via USB, try replacing the power supply with a different one.

Initially, the D1 mini was connected with an external stabilized power supply (via 5V pin).
I tried replacing the power supply with a USB type power supply but the problem persists.
I also tried to change the D1 mini with a new one, but I still have the same problem.
Today I give it a try by disabling the 5G wifi network, but I have many other devices (including D1 mini) connected, which never have data connection problems.

Laxilef, 2025-01-08:
Guys, I'm not saying there is no problem. But I don't know what causes the problem and how to fix it because I can't reproduce it.

Symon84, 2025-01-08:
Ok, I’ve run all the tests I could, including using version 1.5.1 and disabling the 5G WiFi network, but I haven’t seen any improvement.
I’ll try setting up an S2 mini to see if things get better.

Sorry for the question, but on the release page, what’s the difference between these two firmware files:

firmware_s2_mini_1.5.1.bin
firmware_s2_mini_1.5.1.factory.bin
Thank you very much!

Daveblanche, 2025-01-08:
Hi thereSorry, I’m travelling for work and don’t have great internet access. I’m making a massive assumption that this plugin is based on esphome?If so, have a look on the esphome site. It covers connection issues whilst using the single wire thermostat recommended in this project. But I may be wrong, obviously. D

Sent from Sky Yahoo Mail for iPhone

Laxilef, 2025-01-08:
> Sorry for the question, but on the release page, what’s the difference between these two firmware files:
> 
> firmware_s2_mini_1.5.1.bin firmware_s2_mini_1.5.1.factory.bin

Factory for flashing via esptool, not factory for OTA.

Symon84, 2025-01-08:
> Maybe you have some other wifi router to compare with it?
> 

Yes, but replacing the router is not a simple operation (I have many devices connected, including 4 D2 mini that have been working without problems for about a year..)
I keep this option as the "final" test..

> 
> Factory for flashing via esptool, not factory for OTA.
> >
Ok! I've tried with an wemos s2 mini, but I have the same problem..
I've also disabled telnet, serial port and log without any improvement..

As soon as I have some free time I will do more tests..

Symon84, 2025-01-09:
> You can just turn on another router, connect the ESP and your computer to it. You do not need to configure Internet access for this router. To test, you don't need to change the router for all devices :)

In this case I will try as soon as possible and I will check through a ping, even if it will not be connected to home Assistant via MQTT. Thanks!

Symon84, 2025-01-10:
Guys, I found the issue in my case.
Even though the Wemos is installed just 2 meters away from the router and the RSSI signal is -54 dBi, the frequent disconnections are due to the placement of the device (near the boiler).
The boiler has a metal casing, which probably interferes with the signal in some way, even though the Wemos is close to it.
I’ll have to modify the Wemos by adding an external antenna.

It didn’t occur to me immediately because I have other D1 Minis inside walls, near mains voltage, and even 8 meters away from the router, which have no issues at all. Probably, the area near the boiler (even though very close to the router) is subject to interference.

In the following image, you can see how moving the D1 changes the situation: the red zone is where it was near the boiler, while even moving it just a few centimeters makes the connection stable enough.
[image]

Laxilef, 2025-01-10:
Well, if you placed the ESP inside the boiler, then you made a shielded box for it 😄 
In fact, there are many electronics in the boiler that can interfere with the ESP if the ESP is located near the boiler electronics. For example, the ignition works from high voltage, which can sometimes even interfere with the bus.

Now I wonder if other users from this issue have the same reason or not.

Laxilef, 2025-01-10:
@SanFable can you try moving the ESP further away from the boiler?

Symon84, 2025-01-10:
> Well, if you placed the ESP inside the boiler, then you made a shielded box for it 😄
>
> In fact, there are many electronics in the boiler that can interfere with the ESP if the ESP is located near [...]

😅 No, the ESP is outside the boiler, about 20 cm away inside a plastic box, but it is probably still too close to the boiler that somehow interferes with wireless communication.

Symon84, 2025-01-15:
I need to revise my previous statement. The connection actually remained stable for almost two days, but then the same issues resurfaced. I tried moving the device closer to the router, but the problem persists. I downgraded to version 1.4.6, but this didn’t bring any improvements.

I’m investigating other possible causes. It seems that the Wemos gradually slows down over time, eventually causing intermittent Wi-Fi disconnections. I’ll share an update if I find a solution to the issue.

Laxilef, 2025-01-16:
> It seems that the Wemos gradually slows down over time, eventually causing intermittent Wi-Fi disconnections.

In case of memory leaks, ESP restarts would occur.

inediblepotato, 2025-01-18:
I appear to have similar stability issues after upgrading from 1.4.6 to 1.5.1 on my DIYless (D1 mini 32). 

I had backed up, upgraded, restored as in the instructions and I deleted the old MQTT device in HA and let it reappear after the upgrade. Then re-added the indoor and outdoor reporting automations, which appear to work. 

HA shows the Opentherm gateway status as problem every few mins, and all other entities unavailable, screenshot attached.
I disabled any sensors that were coming back as failed-to-receive in the logs (pressure, exhaust temp, etc.), though the failed to receive exhaust temp, DHW temp and flow rate were also present in 1.4.6.  I also tried disabling telnet temporarily but did not observe any improvement in stability.

I saved some logs from telnet over a few mins during problem status moments but I dont see anything revealing. I attach them too. 

To make sure it wasn't a WiFi strength connection issue, I restarted the router, and tried another model (both openWRT), re positioned the device away from anything, it was already well positioned and close to the router. But nothing has improved the stability. No other devices have connection issues.  

I've had to revert to 1.4.6 to keep the house warm, which seems to be working fine again.
Very grateful for the software, been working for months without issue, so please let me know if I can help or provide or do any testing. 

[image]

[OpenThermTelnetLog.txt](https://github.com/user-attachments/files/18465189/OpenThermTelnetLog.txt)

Laxilef, 2025-01-18:
Hi @inediblepotato 

Thanks for the detailed information. I looked at the log and don't see any problems, so I have a few questions:
1. Does the device lose connection to wifi?
2. Does the web interface work when entities are "unavailable" in HA?
3. Are there reboots caused by watchdog?

Symon84, 2025-01-19:
I think I’ve solved it in my case.
It wasn’t just an issue of the Wemos D1 Mini being too close to the boiler, but also a problem with the Wi-Fi network.
In some cases, I had an X-Sense smoke detector, which didn’t have a static IP assigned, randomly taking the IP address of the Wemos where I have the OpenTherm Gateway installed.
After fixing the IP addresses, the system started to remain stable!
I cared a lot about this because I really love this project!
Thank you.

[image]

SanFable, 2025-01-19:
@Symon84 log it for few days. In my case I could have few days ok then one random problem once a day.

I have static ip assigned.

Unfortunately I couldn't move away esp because it's glued :c

Symon84, 2025-01-19:
> log it for few days

Certainly, I’ll let you know in a few days if everything is going well.

Symon84, 2025-01-24:
@SanFable at the moment I have had no problems for 80 hours.
[image]
The version installed is v1.5.1

inediblepotato, 2025-01-24:
Hi @Laxilef ,

I got a chance to test the 1.5.2 release today and tried to get answers to your questions. 
I did the same procedure, deleting the old device in HA, upgrading, restoring settings, re-enabling sensors updates. 
The system was similarly unstable as before. I attach a similar HA screenshot. 
I recorded telnet logs and ping logs for half an hour roughly but again nothing illuminating. 

1. No the device doesn't lose connection to wifi, ping logs shows that but also in router logs, no disconnects.
2. The web interface works during the whole period. It was responsive, no issues.
3. There were no unprompted reboots that I could see in the logs. I changed the verbosity of the log output half way but still nothing indicating a problem.
4. I'm using an up to date eclipse-mosquitto MQTT server running in a docker image. Not much config change from vanilla. Anonymous messages are permitted so I can debug things. I see in the logs that many clients exceed timeout and disconnect only to be directly connected again, opentherm gateway included. But I see these messages when opentherm 1.4.6 is running also. This timeout duration is set by the client as far as I know, not by the server. No other MQTT device on the networks appears to have stability issues. I include an example of the grepped mosquitto logs and my mosquitto.config.

The mosquitto docker stats show no issues, its not taxed, and other than the frequent disconnects from timeouts no other issues in the MQTT logs.

I tried the firmware_d1_mini32_1.5.2.bin, and the firmware_d1_mini32_1.5.2.factory.bin but same behaviour. The attached logs are from the firmware_d1_mini32_1.5.2.bin firmware.

For completeness of info, the device has a static IP set by the router. The traffic on the rest of the network during the testing period is low. 

I reverted back to 1.4.6 and all is working well again. Let me know if I can collect anything else. 

[image]

[image]

[image]

[OpenThermTelnetLog2.txt](https://github.com/user-attachments/files/18539308/OpenThermTelnetLog2.txt)
[pingLog.txt](https://github.com/user-attachments/files/18539309/pingLog.txt)

Laxilef, 2025-01-24:
Hi @inediblepotato 
Thanks. Yes, it seems to have something to do with mqtt keep alive, although it should send a ping packet to keep the connection alive. Since LWT (Last Will and Testament) is triggered when disabled, all entities are marked as unavailable.

It's not very clear why ping doesn't work, but we can try increasing the timeout. Can you compile the firmware from the master branch and test it?

Symon84, 2025-02-05:
At certain times of the day, I still experience peaks of continuous disconnections of the device (even with version 1.5.3).
After analyzing the Home Assistant Core log, it appears that the disconnection times align with the following error:

`Registratore: homeassistant.helpers.template
Fonte: helpers/template.py:821
Prima occorrenza: 12:39:04 (106 occorrenze)
Ultimo accesso: 15:42:02

Error parsing value: 'value_json' is undefined (value: {"heating":{"enabled":true,"turbo":false,"target":20.5,"hysteresis":0.5,"turboFactor":7.5,"minTemp":50,"maxTemp":65},"dhw":{"enabled":true,"target":50,"minTemp":35,""enabled":true,"p_factor":4.5,"i_factor":0.07,"d_factor":0.3,"dt":1800,"minTemp":-10,"maxTemp":10,"deadband":{"enabled":true,"p_multiplier":1,"i_multiplier":0.07,"d_multiplier":1,"thresholdHigh":0.5,"thresholdLow":0.5}}}1Ropentherm/sensors/outdoor_temp{"connected":true,"signalQuality":100,"value":9.5}1Rope, template: {{ value_json.heating.minTemp|float(0)|round(1) }})
Error parsing value: 'value_json' is undefined (value: {"heating":{"enabled":true,"turbo":false,"target":20.5,"hysteresis":0.5,"turboFactor":7.5,"minTemp":50,"maxTemp":65},"dhw":{"enabled":true,"target":50,"minTemp":35,""enabled":true,"p_factor":4.5,"i_factor":0.07,"d_factor":0.3,"dt":1800,"minTemp":-10,"maxTemp":10,"deadband":{"enabled":true,"p_multiplier":1,"i_multiplier":0.07,"d_multiplier":1,"thresholdHigh":0.5,"thresholdLow":0.5}}}1Ropentherm/sensors/outdoor_temp{"connected":true,"signalQuality":100,"value":9.5}1Rope, template: {{ value_json.pid.p_factor|float(0)|round(3) }})
Error parsing value: 'value_json' is undefined (value: {"heating":{"enabled":true,"turbo":false,"target":20.5,"hysteresis":0.5,"turboFactor":7.5,"minTemp":50,"maxTemp":65},"dhw":{"enabled":true,"target":50,"minTemp":35,""enabled":true,"p_factor":4.5,"i_factor":0.07,"d_factor":0.3,"dt":1800,"minTemp":-10,"maxTemp":10,"deadband":{"enabled":true,"p_multiplier":1,"i_multiplier":0.07,"d_multiplier":1,"thresholdHigh":0.5,"thresholdLow":0.5}}}1Ropentherm/sensors/outdoor_temp{"connected":true,"signalQuality":100,"value":9.5}1Rope, template: {{ value_json.pid.dt|int(0) }})
Error parsing value: 'value_json' is undefined (value: {"heating":{"enabled":true,"turbo":false,"target":20.5,"hysteresis":0.5,"turboFactor":7.5,"minTemp":50,"maxTemp":65},"dhw":{"enabled":true,"target":50,"minTemp":35,""enabled":true,"p_factor":4.5,"i_factor":0.07,"d_factor":0.3,"dt":1800,"minTemp":-10,"maxTemp":10,"deadband":{"enabled":true,"p_multiplier":1,"i_multiplier":0.07,"d_multiplier":1,"thresholdHigh":0.5,"thresholdLow":0.5}}}1Ropentherm/sensors/outdoor_temp{"connected":true,"signalQuality":100,"value":9.5}1Rope, template: {{ value_json.pid.maxTemp|float(0)|round(1) }})
Error parsing value: 'value_json' is undefined (value: {"heating":{"enabled":true,"turbo":false,"target":20.5,"hysteresis":0.5,"turboFactor":7.5,"minTemp":50,"maxTemp":65},"dhw":{"enabled":true,"target":50,"minTemp":35,""enabled":true,"p_factor":4.5,"i_factor":0.07,"d_factor":0.3,"dt":1800,"minTemp":-10,"maxTemp":10,"deadband":{"enabled":true,"p_multiplier":1,"i_multiplier":0.07,"d_multiplier":1,"thresholdHigh":0.5,"thresholdLow":0.5}}}1Ropentherm/sensors/outdoor_temp{"connected":true,"signalQuality":100,"value":9.5}1Rope, template: {{ value_json.equitherm.enabled }})`

[image]

It seems there is a communication issue over MQTT from the device to the broker.
I noticed some strange characters in the JSON string.
Could this be the issue?

Symon84, 2025-02-23:
I solved the problem by disabling "WMM Automatic Power Save Delivery(APSD)" in the router! Now the connection is stable and the wifi signal strength is increased.

SanFable, 2025-03-02:
Well, I just have checked my status over 2 weeks and I haven't noticed single fail.

I haven't changed its place or touched anything. I think that esp32s3 is that powerful to handle that?

I'm still on 1.5.0 and don't want to touch anything :D

inediblepotato, 2025-03-09:
Hi Folks, just to follow up. I had compiled the master branch to test the new keepalive a few times but was still seeing issues. 

Given that so many of the other problems seemed to have been caused by WiFi or connection issues I started collecting packets with wireshark and narrow down the cause of my issues. In my case it turns out while the issue got worse when upgrading to the newer version of OTGateway, there was already an issue with all my other MQTT devices reconnecting or being briefly unavailable which I could only see periodically and when i started looking at the MQTT messages themselves. 

I pulled my hair out for a few weeks trying to get to the bottom of it and in my case the problem was caused by EEE (Energy Efficient Ethernet) being enabled on the port of the intel NUC where my MQTT broker docker (and home assistant etc) is running. This would not cause the connection to break enough to reconnect and cause a loggable event in the router or be significantly visible by a long running ping test but would cause frequent but tiny periods of unavailability for the host and therefore the MQTT broker. If MQTT keepalive messages occurred during one of these periods, they would be lost and the MQTT connection would timeout, this also meant that the reconnects/failures on different devices did not occur at the same moment, making it harder to pin down.

I've run into this issue before that linux NIC drivers do not handle EEE well. This issue began when I upgraded the proxmox kernel on my NUC which obviously included updated kernel drivers for the NIC. Once EEE was disabled, no more MQTT reconnects for any device and also for OTGateway no more periods of instability on the new version (1.5.4) . 

I'm uncertain why upgrading the OTGateway version made the problem worse, and reverting improved it (maybe the MQTT messages were more chatty, or just unlucky timeout settings), but happy to have figured it out. Thanks again for all the great work on OTGateway!

closed 2025-06-02 by github-actions[bot]