Laxilef/OTGateway #116: Maximum temperature settings bug(s) in UI

culdaraul, 2024-12-29:
The newly set temperature is not saved after it is changed in UI. 

I think initially after flashing, it was changed after the 1st attempt. Afterwards, it cannot be changed anymore. After a few (dummy) retries, sometimes is changed, sometimes not. Also, changing it via MQTT from Home Assistant has no effect.

HW Setup: ESP32 D1 Mini w/ Opentherm D1 shield. Opentherm comm is established. Indoor/Outdoor sensors fed via MQTT from HomeAssistant via the provided Blueprint in the docs.

There are several bugs, mainly in the UI:
- if there's a sensor configured via GPIO, even if it is disabled, you cannot configure the OpenTherm IN/OUT ports to that GPIO
- for some reason, the "RX Led" in the Opentherm settings had the same GPIO default value as the Rx PIN. Changing it to "blank" / not used is not taken into consideration unless HW Reset is performed. Checked with scope, that Rx PIN had a strong Pull-Down configuration before issuing the reset.
- Switching between Fahrenheit and Celsius would mess all the temperature settings. I ended-up with a hysteresis of 15 and min temperature with "minus"
- Changing the max temp obviously doesn't work. Somehow after a few tens of retries, I could change from 21 to 38 deg C but I'm not able to set it to 45.

Laxilef, 2024-12-30:
Hi,

> The newly set temperature is not saved after it is changed in UI.
> 
> I think initially after flashing, it was changed after the 1st attempt. Afterwards, it cannot be changed anymore. After a few (dummy) retries, sometimes is changed, sometimes not. Also, changing it via MQTT from Home Assistant has no effect.
> ...
> Changing the max temp obviously doesn't work. Somehow after a few tens of retries, I could change from 21 to 38 deg C but I'm not able to set it to 45.

Can you upload the video? I don't quite understand.

> Changing it to "blank" / not used is not taken into consideration unless HW Reset is performed. Checked with scope, that Rx PIN had a strong Pull-Down configuration before issuing the reset.

Thanks, I'll check it.

> if there's a sensor configured via GPIO, even if it is disabled, you cannot configure the OpenTherm IN/OUT ports to that GPIO

The project code does not check for the used GPIOs, this is the responsibility of the user. But there is a validation of the GPIO. If you try to set a non-existent GPIO, the system will not accept these settings.

Laxilef, 2024-12-31:
> * for some reason, the "RX Led" in the Opentherm settings had the same GPIO default value as the Rx PIN. Changing it to "blank" / not used is not taken into consideration unless HW Reset is performed. Checked with scope, that Rx PIN had a strong Pull-Down configuration before issuing the reset.

[image]

culdaraul, 2025-01-02:
Turns out that the issue with the "maximum temperature" was due to the option named "Get min/max temp from boiler" which was by default activated after I flashed the board. Solved for now.

Symon84, 2025-01-19:
I’d like to add that in the latest version 1.5.1, it’s not possible to manually enter a negative temperature for the PID minimum temperature, but if I copy and paste it from text, it works.
Thank you.

[image]

Laxilef, 2025-01-20:
Hi @Symon84 
Test it with the master branch. Thx.

Symon84, 2025-01-28:
> Hi @Symon84 
> Test it with the master branch. Thx.

Ok, in this case, the possibility of inserting the comma appeared, but not the minus symbol. In any case, through the Home Assistant interface there are no problems..

[image]

closed 2025-02-17 by github-actions[bot]