Laxilef/OTGateway #117: Error 00 (diagnostics)

Symon84, 2024-12-30:
Hi, congratulations on the excellent project!
I’m writing regarding a small issue:
I have a Baxi Duo-Tec 25 boiler.
I’m receiving an error code 00 (0x00) from the diagnostics.
The gateway is installed on an ESP8266 (Wemos D1 Mini clone).
The system communicates with Home Assistant via MQTT.
Everything seems to be working fine anyway.

I also have a small request:
I would like to create a room thermostat within Home Assistant that takes advantage of the system’s features. However, I am unable to configure it to display the room temperature (actual and setpoint) instead of the flow water temperature of the system.
[image]
Thank you very much!

Laxilef, 2024-12-30:
Hi,

Pls show a screenshot

Laxilef, 2024-12-30:
Thanks. Yes, it seems to be a trick of your boiler (although my Duo-Tec does not have this).
Usually the diagnostic flag does not affect anything, but sometimes the diagnostic code can be a fault code.

> Heating blocked (I think it’s normal when the room reaches the set temperature)

Yes

Symon84, 2024-12-30:
OK, thank you! I thought it was due to the power set because changing it sometimes the error disappears and it doesn't come back for hours.. I configured MAX 24 kW and MIN 3.5 kW, is that correct? I attach a photo of the boiler label:

[image]

Laxilef, 2024-12-30:
For condensing boilers, I prefer to calculate the average value (of 80/60 and 50/30).

Min power: (3.4 + 3.7) / 2 = 3.55 kW
Max power: (20 + 21.8) / 2 = 20.9 kW

Symon84, 2024-12-30:
Ok! Thanks!!!
But there's a little bug in the UI:
[image]

In the form, it is not possible to enter a number with double precision (it’s not a big issue).
If I access the interface using an iPhone (Safari), it’s not possible to enter numbers with a decimal point (the keyboard doesn’t show it); the only way is to write it somewhere else and copy-paste it.
Sorry for mentioning it here even though it’s unrelated to the post’s title.

Laxilef, 2024-12-30:
> In the form, it is not possible to enter a number with double precision (it’s not a big issue).
> If I access the interface using an iPhone (Safari), it’s not possible to enter numbers with a decimal point (the keyboard doesn’t show it); the only way is to write it somewhere else and copy-paste it.

Thanks, fixed

Symon84, 2024-12-31:
> Thanks, fixed

🔝 thanks!! 🤗

closed 2025-01-08 by Symon84