Laxilef/OTGateway #118: Make switch possible

kuqqum, 2024-12-30:
It woud be great if it can be made possible to select as purpose "Switch"   and as type "on/off" and make this available as entity in HA.
[image]

Daveblanche, 2024-12-30:
To switch heating on and off? It is already available. 
Switch.opentherm_heating/turbo/pid/equitherm are there in mqtt for you to play with.

unless I misunderstand you?

kuqqum, 2024-12-31:
Yes you missed the clou!

Daveblanche, 2024-12-31:
Hmmm

kuqqum, 2024-12-31:
To independently manage the vast and complex world of available (energy) systems, regardless of the manufacturer, there is insufficient control equipment available.
When a basic energy solution is implemented and a specific manufacturer is chosen, it is often logical to continue expanding with the same manufacturer. However, not every manufacturer can provide the most optimal solution for every situation. This frequently results in a combination of equipment from different manufacturers, which are not easily integrated due to inadequate or non-functional communication between control systems. Examples include heating/hot water systems, solar energy installations, home batteries, charging systems for electric vehicles, and energy management systems.

Home Assistant is a platform that can bring all these elements together into a unified and coherent system.
Unfortunately, good interfaces such as OpenTherm Gateway are often lacking. Moreover, there is a need not only for sensors but also for switches that can turn elements on or off. Hence my request.

Laxilef, 2024-12-31:
I recommend you to look at zigbee relay and esphome to solve this problem. Inside Home Assistant you can control different devices and create automations with them.

Trying to implement all possible functionality in one device is not a good idea. In the project, I try to implement only those features that relate to heating.

closed 2025-01-20 by github-actions[bot]