Laxilef/OTGateway #119: Logging changes don't persist across restarts

tincanpete, 2024-12-31:
Hi, while experimenting with both 1.5.0 and 1.5.1 trying to resolve #115 , I have noticed another issue.

Changing the following options in the UI:
* enable serial port
* enable telnet
* Log level 
* and possibly others, I haven't tested

The options save OK and if the page is refreshed, the new values are shown.

However, when the board is restarted, the have gone back to the original, earlier values. 

I wonder if this issue is related to #116 possibly since that one was also reporting "newly set temperature is not saved after it is changed in UI".

Thanks
Pete

Laxilef, 2024-12-31:
Hi,

> However, when the board is restarted, the have gone back to the original, earlier values.

How long after saving do you restart the ESP and how?
Settings are saved in EEPROM no more than once every 60 seconds.

tincanpete, 2024-12-31:
Oh that's interesting, no I wasn't waiting that long. I was trying to re-enable Telnet when I discovered the issue which says it needs a restart, so did that right away. Restart by disconnect USB power and reconnect.  I'll try again after waiting and will let you know if the issue can be closed. Thanks!

Laxilef, 2024-12-31:
You can restart the ESP with a button from the web, then the settings will be saved before restarting

tincanpete, 2025-01-01:
OK there's no issue here after all, I've tested as you suggest and everything is saved OK. Thanks for your help.

closed 2025-01-01 by tincanpete