Laxilef/OTGateway #120: Can not build ESP32 in platformio

Phunkafizer, 2025-01-08:
I can not build the project for all ESP32 targets. ESP8266 targets are building fine.
Tried on two different machines running Win11 with VS code. Other projects with ESP32 build just fine.

********************************************************************************
If you like PlatformIO, please:
- star it on GitHub > https://github.com/platformio/platformio-core
- follow us on LinkedIn to stay up-to-date on the latest project news > https://www.linkedin.com/company/platformio/
********************************************************************************

Warning! Ignore unknown configuration option `version` in section [env]
Warning! Ignore unknown configuration option `version` in section [env]
Resolving esp32_c6 dependencies...
Tool Manager: Installing https://github.com/espressif/arduino-esp32/releases/download/3.1.0/esp32-3.1.0.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32@3.1.0 has been installed!
Tool Manager: Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.3/esp32-arduino-libs-idf-release_v5.3-083aad99-v2.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0%Unpacking...
FileNotFoundError: Traceback (most recent call last):
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 37, in unpack
    return fu.unpack(dst, with_progress=with_progress)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\unpack.py", line 185, in unpack
    self._archiver.extract_item(item, dest_dir)
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\unpack.py", line 49, in extract_item
    self._afo.extract(item, dest_dir)
  File "C:\Users\post\.platformio\python3\Lib\zipfile.py", line 1664, in extract
    return self._extract_member(member, path, pwd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\python3\Lib\zipfile.py", line 1735, in _extract_member
    open(targetpath, "wb") as target:
    ^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'c:\\Users\\post\\Desktop\\OTGateway\\.pio\\.cache\\tmp\\pkg-installing-ggasz_jk\\esp32-arduino-libs\\esp32\\include\\espressif__esp_matter\\connectedhomeip\\connectedhomeip\\src\\app\\clusters\\boolean-state-configuration-server\\BooleanStateConfigurationTestEventTriggerHandler.h'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\project\commands\init.py", line 106, in project_init_cmd
    install_project_dependencies(
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 107, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 149, in _install_project_env_platform
    PlatformPackageManager().install(
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\manager\platform.py", line 62, in install
    p.install_required_packages(force=force)
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 76, in install_required_packages
    self.install_package(name, force=force)
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 70, in install_package
    return self.pm.install(spec or self.get_package_spec(name), force=force)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 47, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_uri(spec.uri, spec)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 191, in install_from_uri
    self.unpack(dl_path, tmp_dir)
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 42, in unpack
    return fu.unpack(dst, with_progress=False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\unpack.py", line 178, in unpack
    self._archiver.extract_item(item, dest_dir)
  File "C:\Users\post\.platformio\penv\Lib\site-packages\platformio\package\unpack.py", line 49, in extract_item
    self._afo.extract(item, dest_dir)
  File "C:\Users\post\.platformio\python3\Lib\zipfile.py", line 1664, in extract
    return self._extract_member(member, path, pwd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\post\.platformio\python3\Lib\zipfile.py", line 1735, in _extract_member
    open(targetpath, "wb") as target:
    ^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'c:\\Users\\post\\Desktop\\OTGateway\\.pio\\.cache\\tmp\\pkg-installing-ggasz_jk\\esp32-arduino-libs\\esp32\\include\\espressif__esp_matter\\connectedhomeip\\connectedhomeip\\src\\app\\clusters\\boolean-state-configuration-server\\BooleanStateConfigurationTestEventTriggerHandler.h'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

Laxilef, 2025-01-08:
Hi,
Try moving the project folder to the root of the disk.
Example: ``C:\OTGateway``

Same problem: https://github.com/pioarduino/platform-espressif32/issues/41

Phunkafizer, 2025-01-08:
This fixed the issue! Thank you very much!

closed 2025-01-08 by Phunkafizer