Laxilef/OTGateway #128: MQTT anonymous connection possible?

x-real-ip, 2025-01-12:
Hi, to connect the opentherm device to a MQTT broker, the username is a required field, is it possible to connect to a broker listener as anonymous, without username and psssword?

Laxilef, 2025-01-12:
Hi,

I think it's a bad idea. Why can this be useful?

x-real-ip, 2025-01-13:
This could be useful for those who have set up an MQTT broker with the `allow_anonymous` option set to `true`. If the broker operates within a secure, isolated environment (e.g., a private network with no external access), this setup could be sufficiently secure. Of course, using a username and password is more secure, but leaving this choice to the user makes the software more flexible and accessible to a wider range of users. For instance, users like me, who configure their setup with the `allow_anonymous` option enabled.

Laxilef, 2025-01-14:
Please build firmware from master branch and test connection to MQTT without credentials

x-real-ip, 2025-01-16:
Thanks so much! it is working without authentication!

From the Eclipse Moquitto logs:
```
1737056318: New connection from 10.42.0.36:52054 on port 1883.
1737056318: New client connected from 10.42.0.36:52054 as opentherm (p2, c1, k15).
```

[image]

[image]

closed 2025-01-16 by x-real-ip