Laxilef/OTGateway #129: Sensor in Add Alternative...

frank-35, 2025-01-17:
Hello Laxilef,
Big thanks and congratulations for sharing this project with us! I have been using your code on a ESP32S with Diyless-shield to control my Intergas boiler for several months now :).

I would like to have these 2 additions:

Firstly, when I open a door or window I would like to prevent the boiler from starting, this would prevent execssive wear on the boiler. This should be easily attainable with a timer to prevent the boiler from starting immediatly.

Secondly, the default return temperature is defined as MsgID=28 in the print summary section, however my Intergas returns this value as AddAlternative MsgID=28. It would be a great improvement if this could be customizable in the sensor settings?

Many thanks for your project so far and keep up the good work.

Greetings,
Frank

Laxilef, 2025-01-18:
Hi,

What is ``AddAlternative MsgID=28``?

> Firstly, when I open a door or window I would like to prevent the boiler from starting, this would prevent execssive wear on the boiler. This should be easily attainable with a timer to prevent the boiler from starting immediatly.

You can create an automation that will turn off the heating when you open the door, and turn it back on after 10 minutes, for example.

frank-35, 2025-01-18:
Hello,
Here the link I found about the return water temperature in the AA-section, hope this helps:

[image]

http://otgw.tclcode.com/firmware.html#configuration

Your project with DS18B20 runs the heater independently, which makes it very reliable (spare ESP32 S3 standby). Interfering in the control-cycle with HA undermines this reliability so a no go for me.

Greetings,
Frank

Laxilef, 2025-01-18:
ID 28 - this is return water temp. This ID is already being polled, so I don't quite understand what the problem is. The boiler may not physically have this sensor, so there is no value for this sensor.

[image]

Laxilef, 2025-01-18:
> Your project with DS18B20 runs the heater independently, which makes it very reliable (spare ESP32 S3 standby). Interfering in the control-cycle with HA undermines this reliability so a no go for me.

Well, you can increase the hysteresis in the heating settings or increase the interval between burner starts in the boiler service menu. I also recommend enabling filtering for the DS18B20 sensor to smooth out sudden temperature changes.
Adding a delay timer is not a good idea.

closed 2025-02-09 by github-actions[bot]