Laxilef/OTGateway #130: Vitodens 100W B1HC-26

michalsimek, 2025-01-18:
Hi,
I have finally made a connection to heater which also have hot water boiler connected. Hot water configuration is working properly and it is also properly reported. Firmware is on 1.5.1. 
But I have an issue with heater itself. Control is working but I have issue with temperature configuration. I have connected 1-wire sensor for indoor temperature (also purpose Indoor temperature) and it is reporting proper values.

I want to setup heater to be active till temperature 20.5 C but in heating tab current temperature is heating temperature not temperature from 1wire sensor. Then my expectation is that based on hysteresis setting temperature will be kept.

[image]

Setup which I would like to have is that when heater is enabled all the time and automation only control target temperature to reach via indoor temperature. And maximum water heating temperature should be around 50 C. 

Here are information about heater:
[image]
And current setup: 
[image]

In main project README.md I see correct values for heating which is reflecting temperature on indoor sensor but not really value from heater water temperature. 

Do you have any idea what's wrong?

Laxilef, 2025-01-18:
Hi,

To control the temperature by air, you must enable and configure Equitherm and/or PID.

michalsimek, 2025-01-18:
That did the trick! Thanks.

closed 2025-01-18 by michalsimek