Laxilef/OTGateway #132: Italian language

bredy73, 2025-01-18:
Hi, I love your project and I translate it in italian. If you want to ad to your next update.

[it.json](https://github.com/user-attachments/files/18465823/it.json)

Laxilef, 2025-01-18:
Hi,

This file is identical to en.json in content. Are you sure you uploaded the right file?

P.S. You can create PR instead of issue.

bredy73, 2025-01-18:
[it.json](https://github.com/user-attachments/files/18466211/it.json)

Laxilef, 2025-01-19:
Added, thanks!

bredy73, 2025-01-19:
thank you for your patience. I'm sorry but I 'm not practice with github

Laxilef, 2025-01-20:
It's not difficult. In the future, you can edit the file directly via Github and then create a PR.

[image]

closed 2025-01-20 by Laxilef