Laxilef/OTGateway #133: DWH startup time

michalsimek, 2025-01-27:
I am using OTgateway for a one week and found one issue with DWH. Output from HA is showing what it is happening. Setup in gateway is minimum temperature 45, maximum 50. That's working properly as is visible around 1am where temperature goes down and then it is heated. 
But we see the issue where water consumption is higher that heater is not able to start and it takes quite a long time to start. 

[image]

Here is a zoom of it. 

[image]

It means reaction time is quite slow. DHW temperature sensor is connected directly to Viessmann 100w that's why heater should be aware about it. Before OTgateway heater was able to keep up much better but I don't have statistics about it. 

What exactly DHW blocking is doing? Is it just disabling hot water or is it a way to let heater to deal with hot water instead of controlling it via OTgateway?

michalsimek, 2025-01-28:
I found this one which is likely the one which shows behavior only for hot water. When I zoom it I see at 22:48:34 hot water is below 45 and at 22:50:53 start heating at that time hot water is at around 37. It is more then 2 mins gap. 

[image]

I will experiment with options to see what they do with my heater.

Laxilef, 2025-01-29:
Hi,
Maximum modulation level in opentherm settings 100%?
I also recommend that you check the boiler service settings. It is possible that the power for heating the DHW is limited there.

michalsimek, 2025-01-30:
Hi, 
yes max modulation level is 100%.
I don't think there is power limitation because I only exchange relay regulator to OTgateway. Nothing else has changed.

[image]

Laxilef, 2025-02-02:
> Setup in gateway is minimum temperature 45, maximum 50.

These parameters affect only the DHW setpoint limits. That is, if you set min=40, max=50, then you will not be able to set the temperature below 40 or above 50.
To set the desired DHW temperature, you must set it in ``climate.opentherm_dhw`` or via the Dashboard page.

So, on the graphs I see that your DHW setting is most likely 50 degrees. The burner turned on when the DHW temperature dropped to ~37 degrees. This means that the hysteresis for turning on the burner for DHW is ~13 degrees. Perhaps this is a lot for your situation, but this parameter can only be configured in the boiler service menu and in no other way.
This is why I recommend that you study the boiler documentation and service menu.

OTGatway does not control the burner in any way and cannot influence the operation of the boiler: we simply transmit the desired temperature to the boiler and it must maintain it.

closed 2025-02-18 by michalsimek