Laxilef/OTGateway #140: The boiler does not turn on even when the temperature drops by 1 degree or more.

Pajuris, 2025-02-09:
Desired temperature is set to 22.5. Now room temperature is 20.9 and boiler is off - not heating.

latest 1.5.3 version. Turbo mode is not working too (can not turn on it - button back to off after few seconds).

Laxilef, 2025-02-14:
Hi,

Show a screenshot of the dashboard page with open tabs.

Pajuris, 2025-02-14:
Sorry, but restored to original firmware (it was cold at home) And it works.
Downgrade to 1.5.1 didn't help either. I think it was hardware problem maybe... A few hours ago, thermostat stop responding at all. Power off/on helped to start device, but boiler was never turned on after that.

Laxilef, 2025-02-14:
Can you test with other hardware (esp/adapter)?

Pajuris, 2025-02-14:
Sorry... i have one Thermostat with WeMos D1 Mini ESP32 and Master OpenTherm Shield only. And I do not use power adapter. Thermostat is connected to Odroid mini PC USB. Now, with original firmware and connected to Odroid USB it works. And it always worked with you firmware.

closed 2025-03-12 by github-actions[bot]