Laxilef/OTGateway #141: Zones support?

dimis7, 2025-02-24:
Hi!

I have a very old Immergas EOLO Star 24 boiler in my home that might be compatible with OT protocol (not sure).

I'm planning to build the circuit to test it, but before i spend my time i want to ask if it's possible to use this project in a multi-zone system.

Currently there are 2x dry contact thermostats in 2 rooms, each controlling an electronic valve.
When the valve if fully open, it is sending a command to the boiler to turn on.

The boiler doesn't know the room temperature. Only know the temperature of the water that is returning from radiators. 
Temperature is set to a fixed value and the boiler will slow down when the water temperature is close to the set point, even if the valves are still open and demanding hot water for the radiators.

Now I'm planning the replace the dry contact thermostats with 2x Sonoff NSPanels, flashed with ESPHome (this firmware: https://github.com/Blackymas/NSPanel_HA_Blueprint)

The original idea was to put the 2x NSPanels as temperature sensors and 1x Sonoff dual WiFi relay and create 2x Generic thermostats in Home Assistant that will control the relays based of the temperature sensors, sending the command to the electronic valves to open / close and the valves will continue to give the command to the boiler to turn on / off.

But it would be nice if i can have the boiler directly connected to home assistant and change the parameters from there, also very useful if i can read the diagnostics (for example I'm using bottles of gas, and when the bottles are empty the boiler is showing error 01 if i remember correctly, meaning ignition failure)

Any ideas are welcome!
Thanks

Laxilef, 2025-02-25:
Hi,

Multi-zones as such are not supported. You can pass the coldest temperature from HA [via automation](https://github.com/Laxilef/OTGateway/wiki#reporting-indooroutdoor-temperature-from-any-home-assistant-sensor) and [turn off the heating](https://github.com/Laxilef/OTGateway/issues/54#issuecomment-1975272298) when all TRVs are closed.

dimis7, 2025-02-27:
Thanks. What about the use of 2x additional GPIO to control the valves via relays? Is this possible?

Laxilef, 2025-02-27:
No, the project does not have such functionality.

closed 2025-03-19 by github-actions[bot]