Laxilef/OTGateway #143: Stability S2 mini vs D1 mini32

pksoft585, 2025-02-26:
Interesting difference in stability on different platforms. Tested on the same device, simple replacement of the board. Same firmware version (latest), only different pin settings due to different GPIO layout.

**S2 mini** - spontaneous reboots from 5, 10 to max. 20 hours.

**D1 mini32** - runs without any failures, Restart only when changing firmware or power failure.

The difference in stability is quite significant. The question is whether only for me or in general. I suppose that this can be caused by the fact that **D1 mini32** has a more powerful processor. According to my experience it is a really well tuned board. 

Please take this as information and not as a complaint. In any case, I would like to thank you for a great job and a perfectly developing product.

Thank you.
Petr

mihsu81, 2025-02-27:
Maybe your board has issues. My S2 mini doesn't reboot randomly.

[image]

pksoft585, 2025-02-27:
Thanks for the information. That is a possible explanation. It's good to know that it works well elsewhere. I don't have an S2 to replace yet and will stick with the D1 mini32.

closed 2025-03-04 by pksoft585