Laxilef/OTGateway #149: Status: INVALID & TIMEOUT

sanderversteegh, 2025-03-15:
I have recently started connecting Home Assistant to my boiler. Flashing and setup was easy and the interface and documentation were also very clear, compliments for that.

My setup is a bit special; I use an ESP32 and DIYLess Master shield but have a Nefit boiler that does not support OpenTherm. That is why I have a conferter that goes from Ems to Opentherm. (https://www.robbshop.nl/nefit-ems-opentherm-converter) 
Unfortunately, the connection from OpenTherm to my boiler is not stable, it only has a connection for seconds after which it loses the connection and it takes a while before it gets the connection back. In the log I get a lot of INVALID and TIMEOUT messages.

`[00:03:36][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:03:36][OT][WARN] Failed receive boiler status: TIMEOUT
[00:03:37][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:03:38][MQTT][TRACE] Successfully publish 844 of 844 bytes to topic: opentherm/state
[00:03:38][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:03:39][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:03:40][SENSORS][NOTICE] #4 'Heating setpoint temp' new value 0: 50.00, compensated: 50.00, raw: 50.00
[00:03:40][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:03:41][MQTT][TRACE] Successfully publish 49 of 49 bytes to topic: opentherm/sensors/heating_setpoint_temp
[00:03:41][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:03:42][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:03:42][OT][WARN] Failed receive boiler status: TIMEOUT
[00:03:43][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:03:44][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  1   Status: INVALID
[00:03:44][OT][WARN] Failed receive boiler status: INVALID
[00:03:44][MQTT][TRACE] Successfully publish 844 of 844 bytes to topic: opentherm/state
[00:03:45][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  1   Status: INVALID
[00:03:45][OT][WARN] Failed receive boiler status: INVALID
[00:03:46][OT][VERB] ID:    0   Request:      300   Response: c0000300   Attempt:  1   Status: SUCCESS
[00:03:46][OT][INFO] Connected, downtime: 226 s.
[00:03:46][OT][INFO] Initializing...
[00:03:46][OT][VERB] ID:  127   Request: 807f0000   Response: 407f0416   Attempt:  1   Status: SUCCESS
[00:03:46][OT][NOTICE] Received slave app version: 22, type: 4
[00:03:47][OT][VERB] ID:  126   Request: 907e0000   Response: f07e0000   Attempt:  1   Status: INVALID
[00:03:47][OT][WARN] Failed set master version
[00:03:47][OT][VERB] ID:  125   Request:   7d0000   Response: f07d0000   Attempt:  1   Status: INVALID
[00:03:47][OT][WARN] Failed receive slave OT version
[00:03:47][SENSORS][DALLAS][TRACE] GPIO 26, polling...
[00:03:47][OT][VERB] ID:  124   Request: 107c0000   Response:        0   Attempt:  1   Status: INVALID
[00:03:47][OT][WARN] Failed set master OT version
[00:03:48][OT][VERB] ID:    3   Request:    30000   Response:        0   Attempt:  1   Status: INVALID
[00:03:48][OT][WARN] Failed receive slave config
[00:03:48][OT][NOTICE] Set master member id: 0, flags: 0
[00:03:48][OT][HEATING][INFO] Enabled
[00:03:48][OT][DHW][INFO] Enabled
[00:03:48][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
[00:03:49][OT][VERB] ID:   15   Request:    f0000   Response:        0   Attempt:  1   Status: TIMEOUT
[00:03:49][SENSORS][DALLAS][TRACE] GPIO 26, sensor #1 'Indoor temp', received data: 19.19
[00:03:49][SENSORS][NOTICE] #1 'Indoor temp' new value 0: 19.19, compensated: 19.19, raw: 19.19
[00:03:49][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:03:49][MQTT][TRACE] Successfully publish 845 of 845 bytes to topic: opentherm/state
[00:03:50][MQTT][TRACE] Successfully publish 53 of 53 bytes to topic: opentherm/sensors/indoor_temp
[00:03:50][OT][VERB] ID:   15   Request:    f0000   Response:        0   Attempt:  2   Status: TIMEOUT
[00:03:50][SENSORS][NOTICE] #4 'Heating setpoint temp' new value 0: 50.00, compensated: 50.00, raw: 50.00
[00:03:50][MQTT][TRACE] Successfully publish 49 of 49 bytes to topic: opentherm/sensors/heating_setpoint_temp
[00:03:51][OT][VERB] ID:   15   Request:    f0000   Response:        0   Attempt:  3   Status: TIMEOUT
[00:03:52][MAIN][VERB] Free heap size: 47220 of 250712 bytes (min: 19952, diff: 0), max free block: 36852 (min: 14836, diff: 0, frag: 21%)
[00:03:52][OT][VERB] ID:   15   Request:    f0000   Response:        0   Attempt:  4   Status: TIMEOUT
[00:03:53][OT][VERB] ID:   15   Request:    f0000   Response:        0   Attempt:  5   Status: INVALID
[00:03:53][OT][WARN] Failed receive min modulation and max power
[00:03:54][OT][VERB] ID:   14   Request: 900e6400   Response:        0   Attempt:  1   Status: INVALID
[00:03:54][OT][WARN] Failed set max modulation: 100% (response: 100%)
[00:03:54][SENSORS][NOTICE] #9 'Modulation level' new status: CONNECTED
[00:03:54][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:03:54][SENSORS][NOTICE] #10 'Power' new status: CONNECTED
[00:03:54][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
[00:03:54][OT][VERB] ID:   25   Request: 80190000   Response: 40193db3   Attempt:  1   Status: SUCCESS
[00:03:54][OT][HEATING][NOTICE] Received temp: 61.70
[00:03:54][SENSORS][NOTICE] #2 'Heating temp' new status: CONNECTED
[00:03:54][SENSORS][NOTICE] #2 'Heating temp' new value 0: 61.70, compensated: 61.70, raw: 61.70
[00:03:54][MQTT][TRACE] Successfully publish 53 of 53 bytes to topic: opentherm/sensors/heating_temp
[00:03:55][MQTT][TRACE] Successfully publish 48 of 48 bytes to topic: opentherm/sensors/modulation_level
[00:03:55][OT][VERB] ID:   28   Request: 801c0000   Response: 401c3699   Attempt:  1   Status: SUCCESS
[00:03:55][OT][HEATING][NOTICE] Received return temp: 54.60 (converted: 54.60)
[00:03:55][SENSORS][NOTICE] #3 'Heating return temp' new status: CONNECTED
[... 20 lines cut ...]
[00:03:58][OT][VERB] ID:   28   Request: 801c0000   Response: 401c3699   Attempt:  1   Status: SUCCESS
[00:03:58][OT][HEATING][NOTICE] Received return temp: 54.60 (converted: 54.60)
[00:03:58][SENSORS][NOTICE] #3 'Heating return temp' new value 0: 54.60, compensated: 54.60, raw: 54.60
[00:03:58][OT][VERB] ID:   56   Request: 10382800   Response: 70382800   Attempt:  1   Status: INVALID
[00:03:58][OT][DHW][WARN] Failed set temp
[00:03:58][MQTT][TRACE] Successfully publish 53 of 53 bytes to topic: opentherm/sensors/heating_temp
[00:03:58][MQTT][TRACE] Successfully publish 53 of 53 bytes to topic: opentherm/sensors/heating_return_temp
[00:04:00][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  1   Status: TIMEOUT
[00:04:00][SENSORS][NOTICE] #4 'Heating setpoint temp' new value 0: 50.00, compensated: 50.00, raw: 50.00
[00:04:01][MQTT][TRACE] Successfully publish 49 of 49 bytes to topic: opentherm/sensors/heating_setpoint_temp
[00:04:01][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  2   Status: TIMEOUT
[00:04:01][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:04:02][MQTT][TRACE] Successfully publish 849 of 849 bytes to topic: opentherm/state
[00:04:02][MAIN][VERB] Free heap size: 44484 of 250712 bytes (min: 18148, diff: 1804), max free block: 36852 (min: 14836, diff: 0, frag: 17%)
[00:04:02][SENSORS][DALLAS][TRACE] GPIO 26, devices on bus: 1, DS18* devices: 1
[00:04:02][SENSORS][DALLAS][TRACE] GPIO 26, polling...
[00:04:02][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  3   Status: TIMEOUT
[00:04:03][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  4   Status: TIMEOUT
[00:04:04][SENSORS][DALLAS][TRACE] GPIO 26, sensor #1 'Indoor temp', received data: 19.25
[00:04:04][SENSORS][NOTICE] #1 'Indoor temp' new value 0: 19.25, compensated: 19.25, raw: 19.25
[00:04:04][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  5   Status: TIMEOUT
[00:04:04][OT][WARN] Failed receive boiler status: TIMEOUT
[00:04:04][OT][WARN] Disconnected, uptime: 18 s.`

If it did not work at all I would understand that, but is there anything I can do to prevent those TIMEOUT's?

Laxilef, 2025-03-15:
Hi @sanderversteegh,

It is difficult to guess why this converter at some point in time stops responding (TIMEOUT - no response within the allowed time).
What is the length of the cable from the converter to DIYLESS? What cable is used? Are there any other power cables nearby?

What board do you use with the DIYLESS adapter?

sanderversteegh, 2025-03-16:
The cable length to DIYLESS to converter is about 5cm and from converter to boiler about 10cm.
I now use a esp32-wroom-32 nodemcu 30pin. Bud i gone try a different board today.

Thank you for you're help!

Laxilef, 2025-03-17:
Pls test with this build

[build.zip](https://github.com/user-attachments/files/19277766/build.zip)

sanderversteegh, 2025-03-17:
I tried the build. It seems a lot better but still flaky. Especially reading data seems to be something it has trouble with. Switching the heating on/off and sending the temperature to the boiler seems to work quite well.

> [03:13:09][OT][VERB] ID:    0   Request:      300   Response: 50000000   Attempt:  1   Status: SUCCESS
> [03:13:09][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 0; cooling: 0; fault: 0; diag: 0
> [03:13:09][SENSORS][NOTICE] #9 'Modulation level' new status: CONNECTED
> [03:13:09][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
> [03:13:09][SENSORS][NOTICE] #10 'Power' new status: CONNECTED
> [03:13:09][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
> [03:13:09][MQTT][TRACE] Successfully publish 48 of 48 bytes to topic: opentherm/sensors/modulation_level
> [03:13:09][OT][VERB] ID:   25   Request: 80190000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:09][OT][HEATING][WARN] Failed receive temp
> [03:13:09][MQTT][TRACE] Successfully publish 48 of 48 bytes to topic: opentherm/sensors/power
> [03:13:09][OT][VERB] ID:   28   Request: 801c0000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:09][OT][HEATING][WARN] Failed receive return temp
> [03:13:10][OT][VERB] ID:   18   Request:   120000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:10][OT][WARN] Failed receive pressure
> [03:13:10][OT][VERB] ID:   56   Request: 10382800   Response:        0   Attempt:  1   Status: INVALID
> [03:13:10][OT][DHW][WARN] Failed set temp
> [03:13:10][OT][VERB] ID:   57   Request: 10393100   Response:        0   Attempt:  1   Status: INVALID
> [03:13:10][OT][HEATING][WARN] Failed set max heating temp: 49.20 (converted: 49.20)
> [03:13:11][OT][VERB] ID:    1   Request: 90013133   Response:        0   Attempt:  1   Status: INVALID
> [03:13:11][OT][HEATING][WARN] Failed set target temp
> [03:13:12][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  1   Status: INVALID
> [03:13:12][OT][WARN] Failed receive boiler status: INVALID
> [03:13:12][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
> [03:13:12][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
> [03:13:12][MQTT][TRACE] Successfully publish 48 of 48 bytes to topic: opentherm/sensors/modulation_level
> [03:13:12][OT][VERB] ID:   25   Request: 80190000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:12][OT][HEATING][WARN] Failed receive temp
> [03:13:12][MQTT][TRACE] Successfully publish 48 of 48 bytes to topic: opentherm/sensors/power
> [03:13:12][OT][VERB] ID:   28   Request: 801c0000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:13][OT][HEATING][WARN] Failed receive return temp
> [03:13:13][OT][VERB] ID:   18   Request:   120000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:13][OT][WARN] Failed receive pressure
> [03:13:13][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
> [03:13:13][OT][VERB] ID:   56   Request: 10382800   Response:        0   Attempt:  1   Status: INVALID
> [03:13:13][OT][DHW][WARN] Failed set temp
> [03:13:13][MQTT][TRACE] Successfully publish 852 of 852 bytes to topic: opentherm/state
> [03:13:13][OT][VERB] ID:   57   Request: 10393100   Response:        0   Attempt:  1   Status: INVALID
> [03:13:14][OT][HEATING][WARN] Failed set max heating temp: 49.20 (converted: 49.20)
> [03:13:14][OT][VERB] ID:    1   Request: 90013133   Response:        0   Attempt:  1   Status: INVALID
> [03:13:14][OT][HEATING][WARN] Failed set target temp
> [03:13:14][SENSORS][NOTICE] #4 'Heating setpoint temp' new value 0: 49.20, compensated: 49.20, raw: 49.20
> [03:13:14][MQTT][TRACE] Successfully publish 51 of 51 bytes to topic: opentherm/sensors/heating_setpoint_temp
> [03:13:15][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  1   Status: INVALID
> [03:13:15][OT][WARN] Failed receive boiler status: INVALID
> [03:13:15][SENSORS][NOTICE] #9 'Modulation level' new value 0: 0.00, compensated: 0.00, raw: 0.00
> [03:13:15][SENSORS][NOTICE] #10 'Power' new value 0: 0.00, compensated: 0.00, raw: 0.00
> [03:13:15][OT][VERB] ID:   25   Request: 80190000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:15][OT][HEATING][WARN] Failed receive temp
> [03:13:15][MQTT][TRACE] Successfully publish 48 of 48 bytes to topic: opentherm/sensors/modulation_level
> [03:13:16][OT][VERB] ID:   28   Request: 801c0000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:16][OT][HEATING][WARN] Failed receive return temp
> [03:13:16][MQTT][TRACE] Successfully publish 48 of 48 bytes to topic: opentherm/sensors/power
> [03:13:16][OT][VERB] ID:   18   Request:   120000   Response:        0   Attempt:  1   Status: INVALID
> [03:13:16][OT][WARN] Failed receive pressure
> [03:13:16][OT][VERB] ID:   56   Request: 10382800   Response:        0   Attempt:  1   Status: INVALID
> [03:13:16][OT][DHW][WARN] Failed set temp
> [03:13:17][OT][VERB] ID:   57   Request: 10393100   Response:        0   Attempt:  1   Status: INVALID
> [03:13:17][OT][HEATING][WARN] Failed set max heating temp: 49.20 (converted: 49.20)
> [03:13:17][OT][VERB] ID:    1   Request: 90013133   Response:        0   Attempt:  1   Status: INVALID
> [03:13:17][OT][HEATING][WARN] Failed set target temp
[... cut ...]
> [03:37:53][MAIN][VERB] Free heap size: 12372 of 250176 bytes (min: 7072, diff: 0), max free block: 4084 (min: 3828, diff: 0, frag: 66%)
> [03:37:54][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  1   Status: INVALID
> [03:37:54][OT][WARN] Failed receive boiler status: INVALID
> [03:37:55][OT][VERB] ID:    0   Request:      300   Response:        0   Attempt:  1   Status: INVALID
> [03:37:55][OT][WARN] Failed receive boiler status: INVALID

I am going to see how reliable writing to the boiler really is. Maybe this is enough.

Laxilef, 2025-03-18:
It's still awful :(
I suspect that this converter is not working correctly. Perhaps you should contact customer service?

closed 2025-04-07 by github-actions[bot]