Laxilef/OTGateway #152: OTGateway keeps heating (it's summer)

dxlbnl, 2025-05-04:
Hi, Thanks for you're awesome project. 

I recently installed the otgateway on my intergas kombikompact hre. And integrated it with Homeassistant. I've setup the blueprint for logging the internal temperature to it. 

Currently it's warm outside, the temperature logged is 22 C, but the heating state stays enabled. In the dashboard control I disabled heating. 
I don't seem to be able to set the desired temperature. 

I'm sure stuff is misconfigured, but I'm usure where to begin

[image]
[image]
[image]

dxlbnl, 2025-05-04:
I noticed a restart due to watchdogs: 

```json
{
  "build": {
    "version": "1.5.4",
    "date": "Mar  6 2025 19:26:16",
    "env": "s3_mini",
    "core": "3.1.3",
    "sdk": "v5.3.2-584-g489d7a2b3a-dirty"
  },
  "heap": {
    "total": 307768,
    "free": 75464,
    "minFree": 2780,
    "maxFreeBlock": 31732,
    "minMaxFreeBlock": 2780
  },
  "chip": {
    "model": "ESP32-S3",
    "rev": 1,
    "cores": 2,
    "freq": 240
  },
  "flash": {
    "size": 4194304,
    "realSize": 4194304
  },
  "crash": {
    "reason": "Reset due to other watchdogs",
    "core": 166,
    "heap": 1058299551,
    "uptime": 151756331
  }
}
```

Laxilef, 2025-05-07:
Hi @dxlbnl,
For equitherm to work, you need to transmit indoor temperature. Since you have selected the ``Manual via MQTT/API`` type, you must configure automation as described [here](https://github.com/Laxilef/OTGateway/wiki#reporting-indooroutdoor-temperature-from-any-home-assistant-sensor).

dxlbnl, 2025-05-07:
I did, I didn't enable equitherm, just the PID controller. So the OTGateway would work as a thermostat. But I'm not sure how to send the desired temperature

Laxilef, 2025-05-07:
For PID it is also necessary to transmit indoor temperature. The link to the instructions is above.

andrey150960, 2025-05-28:
[dxlbnl](https://github.com/dxlbnl)
How was the esp 32 s3 flashed? I can't do it

Laxilef, 2025-06-09:
> Though there is no way to set a proper target temp, so I'm unsure what value it uses as target. I'd expect HA also to be able to set target temp too.

Why? Use ``climate.opentherm_heating`` to set the desired indoor temp

closed 2025-06-29 by github-actions[bot]