Laxilef/OTGateway #153: GPIO pin access through MQTT?

yellius, 2025-05-10:
Great project! I'm using it perfectly with a Remeha Calenta gas heater.

Feature request: could one or more of the unused GPIO pins be configured as input or output and exposed as a senor or switch in HA?
This would be great for some custom (hardware) hacking and optimisation.

Laxilef, 2025-05-12:
Hi,

Thanks! You can connect additional sensors (NTC, Dallas) to free compatible GPIOs.
But there is no support for additional switches. What can this be useful for?

yellius, 2025-05-12:
Hi, I was thinking of using it to control a three-way water valve to switch
tapwater between electric boiler and gas boiler, depending on electric
boiler temperature.

yellius, 2025-05-13:
> I think that unnecessary functionality can complicate the project, so I
> don’t want to do this. To control a powerful load, you will still need a
> relay, and maybe more than one. Kincony + esphome or similar inexpensive
> devices can do this perfectly.

Ok - thanks for having considered!

closed 2025-06-02 by github-actions[bot]