Laxilef/OTGateway #155: Heat and DHW constantly off but burner sometimes runs

kriskristoferson, 2025-05-31:
I have a gateway connected to a Vitodens 100 boiler (an older model from around 2011). Sometimes, for completely unknown reasons, it starts periodically activating the burner, even though both Heat and DHW are turned off the entire time. Occasionally, it manages to stay off for a longer period (possibly after a restart or shutdown, though I'm not sure how or why), but then it returns to this periodic heating again. I have both indoor and outdoor temperature sensors connected via MQTT to Home Assistant. 
I’m attaching a burner modulation graph (Heat and DHW remain off the whole time).

[image]

Laxilef, 2025-06-03:
I see that the setpoint temp is always around 20 degrees. Are you sure that your boiler correctly accepts this temperature? What is the minimum temperature you can set from the boiler panel?

kriskristoferson, 2025-06-05:
Short logs with pid and equiherm off
[pid_equitherm_off.txt](https://github.com/user-attachments/files/20603335/pid_equitherm_off.txt)

Laxilef, 2025-06-05:
Ok, thanks
Could you upload longer logs that would include the moments when the burner was turned on?

> 1. Your boiler does not actually accept the 20 degree setpoint, and the minimum temperature for it is ~40 degrees. But this requires confirmation: either from the boiler documentation (but for this you need the exact model name), or by indirect signs: indication on the boiler panel, etc. For this reason, the boiler heats up to ~40 degrees.

Not confirmed:
```[00:13:00][OT][HEATING][NOTICE] Received min temp: 20 (converted: 20), max temp: 80 (converted: 80)```

Laxilef, 2025-06-06:
If the heating is off, it should not start even when using pid/equitherm.

kriskristoferson, 2025-06-09:
About 15 minutes, during which time the heating was activated; unfortunately, the clock on the module could not be set via NTP.
[otgw.zip](https://github.com/user-attachments/files/20657649/otgw.zip)

Laxilef, 2025-06-13:
Thanks

This is very unusual:
```
[00:03:54][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 1; cooling: 0; fault: 0; diag: 0
...
[00:04:31][OT][NOTICE] Received boiler status. Heating: 0; DHW: 0; flame: 1; cooling: 0; fault: 0; diag: 0
```

Even though we turned off the heating and DHW, the boiler still ignites the burner...
Can you try changing the options in opentherm settings and test again?
``Summer/winter mode`` = off
``Heating state as summer/winter mode`` = on

Laxilef, 2025-06-28:
Or try with ``Duplicate heating to CH2``

Laxilef, 2025-07-03:
Can you take a screenshot of your opentherm settings and upload the log again, which will show the moments when the burner was turned on?
I think the burner may be turning on because channel 2 is on.

kriskristoferson, 2025-07-03:
These are the current settings and logs from the burner start. I was hoping that since outside the heating season the PID and equitherm are unnecessary, I would try turning them off, but it didn’t help. In any case, PID is currently off and equitherm is off.

[otgw3.zip](https://github.com/user-attachments/files/21032437/otgw3.zip)

[image]

Laxilef, 2025-07-03:
The log looks normal, channel 2 is disabled. But the boiler still turns on the burner, I have no idea why this happens :(

I read in the instructions for your boiler that when connecting an opentherm thermostat, jumper D must be removed. Have you done this?
[image]

kriskristoferson, 2025-07-04:
Hmmm, that’s strange — my manual looks like this. I had jumper D on X9, I’ve removed it now, so we’ll see.
[image]

kriskristoferson, 2025-07-07:
[image]

I interpreted it as meaning that instead of the jumper on X9, you connect an on/off thermostat — but maybe it really should be understood in the context of Opentherm thermostats. Anyway, the important thing is that the issue seems to be resolved. I’ve already seen at least two temperature levels where previously the burner would have definitely turned on, and nothing happened this time. If it gets worse again, I’ll let you know. For now, everything is fine — the domestic hot water heating works, and I can patiently wait for the heating season to test the central heating temperature control.

kriskristoferson, 2025-07-19:
There’s still some issue. I hadn’t touched the boiler for a few days — the burner only operated when I turned on DHW. Yesterday, there was a brief power outage, and a few hours later I noticed the burner was again caught in a startup loop. I briefly turned off the power to the boiler and the gateway. That seemed to help. The jumper is still removed.
So it seems once again like some strange behavior related to powering on the boiler and the module.
[image]

Laxilef, 2025-07-19:
Damn. I hoped the problem was solved...
Were there any faults on the boiler panel? Or did it just start firing the burner?

kriskristoferson, 2025-07-21:
No, I don't think so. I didn't check at the time, but the boiler was serviced a month ago and I didn't notice any issues or error messages. It was just a brief power outage in the whole house after which the problem appeared. When I noticed it, I turned the boiler and the module off and on again — the problem disappeared and so far everything is fine. Other than that, I haven’t touched either the boiler or the module configuration for several days.

koodergit, 2025-08-06:
Hi, not sure if this is the correct place to post this but I have exactly the same boiler and of the same vintage (2011) and I too get this problem. I need to collect the logs like you have done already but I didn't want the issue to go stale if indeed it is an issue and not just a settings thing.   

Happy to help and support as needed.

koodergit, 2025-08-08:
These settings for me completely remove the issue you are seeing. 

[image]

The boiler does not fire unless directed by the automations which is exactly as it should be. 

Note that I still have priority dhw with a separate thermostat but this does not interfere with the open therm commands as long as I turn it below the point of OTG.

Maybe give them a try and see if it fixes your issue.

kriskristoferson, 2025-08-09:
For me, the problem comes back every few days, usually after a power outage at home. It seems the key point is when both devices start up together. That’s what happened the day before yesterday, I think — I noticed that for several hours the burner was switching on and off cyclically. I turned the boiler off for a moment (the module remained on), and the problem disappeared.

Laxilef, 2025-08-10:
If the problem occurs when the boiler and ESP are turned off, use an inexpensive UPS.
I have no idea why the boiler behaves this way and I have not seen such behavior on other boilers.

closed 2025-08-30 by github-actions[bot]