Laxilef/OTGateway #157: Emergency mode starts even when Heating is disabled

mihsu81, 2025-06-10:
Hi @Laxilef ,

Emergency mode starts even though Heating is disabled. It should consider whether Heating is actually enabled or not.
Would it be possible to make this adjustment?

Thank you in advance.

[image]

[image]

[image]

Laxilef, 2025-06-13:
Hi,

This is normal behavior.

Laxilef, 2025-06-13:
It may seem strange, but an emergency mode can prevent your home from freezing someday. Just add additional sensors (wired ones are possible).

mihsu81, 2025-06-13:
I'm sorry if I wasn't clear. I'm not debating the usefulness of Emergency mode. It is definitely beneficial.
What looks like an issue, to me at least, is that during the summer, when I turn off the heating, the thermostat shouldn't start heating the house just because there was a communication issue with Home Assistant and it went into Emergency mode.
I expect the Emergency Mode algorithm to check if the user has already turned off the heating before turning it on.

Laxilef, 2025-06-16:
Ok, I'll think about it.

ZHcom, 2025-07-10:
> Slightly changed the logic of the emergency mode. Now he doesn't turn on the heating.
> 
> However, there are now freeze protection settings: now the heating will be automatically turned on if the temperature of the heat carrier/indoor temperature drops below the set threshold (`Heating settings` > `Freeze protection` > `Low temperature threshold`).

**where can I download a new version of the firmware to enable this function or how to enable it on version 1.5.5?**

ZHcom, 2025-09-26:
Good afternoon, I installed the new firmware 1.5.6 but didn't see where to enable or disable emergency mode. Where can I find this function? Or how does it work?