Laxilef/OTGateway #160: Viessman vitopend 100-W

RastopchinSS, 2025-06-18:
Hello!

I have a Viessman vitopend 100-W A1HB-34 gas boiler installed.
I have connected an OpenTherm controller with firmware 1.5.5
The boiler detects an OpenTherm connection. 
In the Web interface, I see readings of the current coolant temperature, the set temperature, the burner status, and the maximum and minimum temperature settings.
But the boiler does not accept the target temperature of the coolant. He sets it at 40 degrees Celsius. 
The burner turns on for a few seconds and turns off before reaching the set temperature on OpenTherm.

Please help me figure out the settings. What settings should I make so that the boiler operates at a given temperature of the coolant?

Laxilef, 2025-06-18:
Hi,

Try with OT option "Duplicate heating to CH2"

Laxilef, 2025-06-19:
Try disabling the "Sync max heating temp with target temp" option, reset the boiler settings and try setting the temperature to 60 degrees

RastopchinSS, 2025-06-21:
I have the following problem. 
I turn off the heating in the controller settings. It is displayed that the heating is turned off. But the heating is active on the boiler and the burner is periodically switched on for a short time. 
I can also see it in the Home Assistant - the heating on the boiler is working.

[image]

[image]

RastopchinSS, 2025-06-22:
> `Heating state as summer/winter mode`
> And `Sync max heating temp with target temp` = off

It really worked. Thank you so much!

closed 2025-07-18 by github-actions[bot]