Laxilef/OTGateway #162: Can I use a Chinese-language interface? How should I operate it?

david2dfs23, 2025-07-02:
I am trying to learn and use this gateway. Your project is really great. I would like to ask: Can I use a Chinese-language interface? How should I operate it?

david2dfs23, 2025-07-04:
[#/](url)

> Thank you very much for your reply, my friend. I'm not very familiar with using this platform. Is this how I should send it? If it's not right, please let me know again.

[cn.json](https://github.com/user-attachments/files/21050477/cn.json)

Laxilef, 2025-07-06:
Hi,
Already added, PR #163

closed 2025-07-06 by Laxilef

david2dfs23, 2025-07-06:
太棒了👍。正是我需要的。非常感谢您🙏

---- Replied Message ----
| From | ***@***.***> |
| Date | 07/06/2025 22:23 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [Laxilef/OTGateway] Can I use a Chinese-language interface? How should I operate it? (Issue #162) |
Laxilef left a comment (Laxilef/OTGateway#162)

Hi,
Already added, PR #163