Laxilef/OTGateway #166: Emergency mode target temperature defaults to 40°C

pieczka-marek, 2025-07-06:
Hello, first of all Laxilef and the dev team. Thank you for the great work and keeping my house warm ;)

The issue is that several times the emergency mode kicked in when my HA service was down which started to heat the home with 40°C water overheating it quickly.

I lowered the emergency target temp to 25°C only to find it back at its default value of 40°C (the temp stays at 25 after saving and for some time, not sure when the defaulting happens). This might be partially fixed by https://github.com/Laxilef/OTGateway/issues/157 but in any conditions the temperature of 40°C will be too much. Seems like a bug that the temp is defaulted to some other value.

System info: 
boiler: Termet Ecocondens Gold Plus 25kW, 
IC: ESP8266 with DIYless shield, 
Version 	1.5.5
Build 	Env: d1_mini
Core: 3.1.2
SDK: 3.0.5(b29dcd3)

[image]

[image]

[image]

pieczka-marek, 2025-07-06:
Laxilef, thank you for your reply.
I have set the `Get min/max temp from boiler ` only recently the issue is independant from that. I investigated a bit more and I think I know what is happening:
1. I have a automation to increase the minimum heating temp to 35°C for a short time until the flame is detected. I need it to fire the boiler at low heating temp settings. Its a kind of hack to override the equitherm setpoint temp.
2. This automation is run after HA restart
3. The emergency temp target is reset to default value when the target temp is higher than the target (makes sense)

Would it be possible to make the emergency target temp configurable from HA? Do you think it would be a reasonable solution for that issue and something that you would consider as a worthy improvement?

Laxilef, 2025-07-06:
Yes, that's probably what happens.

> Would it be possible to make the emergency target temp configurable from HA?

In previous versions there was such a possibility. But I deleted some entities, including emergency target temp, because there were too many entities.

Not very elegant, but working solution for HA:
```
action: mqtt.publish
data:
  topic: opentherm/settings/set
  payload: |-
    {
      "emergency": {
        "target": 25
      }
    }

```

closed 2025-07-07 by pieczka-marek