Laxilef/OTGateway #171: Wifi connect hang every second start

mipisoft, 2025-09-03:
During every second start wifi cant connect. On first start is everything OK, When lost power supply and then back wifi cant connect and OTGateway working, but no access. Next start is OK
I use D1_Mini chinese version 2 pieces  OtGateway version 1.5.5
Comunicate with Atlantic Loria heat pump  working flawlessy but only basic sensors. DHW - no problem  Heating still not tested.

Laxilef, 2025-09-28:
You can save logs by connecting the ESP via USB cable and connecting to the serial port using PuTTY, for example.

djp3k05, 2025-09-28:
Got a similar issues, with the reconect to the wifi network. The wifi signal is not very good in the room where the wemos d1 mini is. About -65dbm. Every few days the device is loosing the wifi connection and will not reconect back by itself. I have to power it off and back on. For now I've added a zigbee power socket for the wemos. If the wemos is not replying to ping for 2 minutes, HA will power off and then power on the zibee power socket.
Can you add/improve the reconnection logic? To try to connect to wifi indefinitely until it succede?
Thank you for your hard work. This is the first year when I Will use your opentherm implementation.

Laxilef, 2025-10-03:
> I compile soft after the laxilef changes.

After what changes?

Laxilef, 2025-10-03:
> After this change: [14826c1](https://github.com/Laxilef/OTGateway/commit/14826c10cd5c71e2dc0e5b2e7645791d5c930c05)

This does not introduce any changes to the interaction with the network.

Try [this](https://github.com/Laxilef/OTGateway/issues/171#issuecomment-3344266128).

djp3k05, 2025-10-04:
> > Can you add/improve the reconnection logic? To try to connect to wifi indefinitely until it succede?
> 
> Already [implemented](https://github.com/Laxilef/OTGateway/blob/master/lib/NetworkUtils/NetworkMgr.h#L290-L378). For some reason, the ESP can't connect to the router. If you use 2G and 5G networks, you must use different names, for example `Name_2G`, `Name_5G`, since not all ESPs support 5G networks.
> 
> In addition, I recommend trying to use only WPA2-Personal and disabling WMM on the router.

I'm using different names for 2.4G and 5G (Asus Router). As I said, my issue is that after few days get's disconnected and the only way is to reboot the Wemos device. Rebooting the router while the wemos is connected - it does reconnect back every time. Rebooting the router while the Wemos lost the connection - does not reconnect back. I also added a 3db gain antenna to the wemos - now I got 100% signal. but still got those disconnects. Maybe the Wemos board is freezing (and not an issue with the OTG networking stack). I got a different Wemos board - so I will test with that in the next days. Currently I got a worse issue, will open a new issue for that.

Laxilef, 2025-10-04:
In my experience, I've seen reboot loops and ESP freezes due to power supply issues. Therefore, I recommended you try a different power supply.