Laxilef/OTGateway #178: Unable to set DHW temp setpoint or maxTemp to above 60°C

pieczka-marek, 2025-09-28:
Hello Laxilef,

second year with your opentherm integration, looking confident into the upcoming heating season! Thank you for all your hard work :D

I have only one issue that has occurred I think about the time of 1.5.0 release. Before I was able to set the DHW temp to 70°C now that value is capped at 60°C.

Why is this a issue? 
I have a anti legionella automation set up which heats the hot water tank to a higher temperature for a short period of time making sure the tank is sterile. Temperature of 60°C is not ideal for that, the recommended range is 70-80°C.

Kindly request to increase the dhw max temp to 70°C.

Regards,
Marek

pieczka-marek, 2025-09-30:
Yeah, so the limits are `DHW limits:   35...65 °C` I guess they are hardcoded in the boiler itself with no possibility to change? Or I can just remove them from OTGateway and set what I want. And also I am able to set DHW max temp to 65°C.

Laxilef, 2025-09-30:
> I guess they are hardcoded in the boiler itself with no possibility to change?

yes

> Or I can just remove them from OTGateway and set what I want.

It's unlikely, the boiler won't accept DHW temp above 65